Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #97: Wake from sleep with SMS

golden-oldie, 2020-11-01:
I'm trying to put the T-Call into light sleep mode and then wake it up when it receives an SMS. The documentation says the ESP32 should support this as well as the SIM800 going into sleep but still able to receive SMS.

Is there any example code out there for this?

toph84, 2020-12-05:
Hi, I'm a complete newbie with programing microcontrollers, but I got it to work with adapted code from here and there. Maybe I can point you in the right direction, I'm afraid I can't give all the credits for different parts of the code...

Using AT-command "AT+CFGRI=1" sets the modem to pulse low the RI pin for ~100ms when SMS is received. You can use this to notify the ESP to wake up from sleep mode, if your version of T-Call has the RI-Pin connected. Remember when going to sleep mode you have to keep modem power PIN high.

```
  gpio_deep_sleep_hold_en();  //don't know why and if it is required 
  delay(10);
  gpio_hold_en(GPIO_NUM_23);  //keep modem power pin high
  delay(10);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_33, 0); //define wakeup source pin and state, 1 = High, 0 = Low
  delay(10);
  esp_light_sleep_start();
```

You can use AT-Command "AT+CSCLK=2" which, according to SIM800 Manual:

> 2. Enables slow clock automatically, when there is no interrupt (on air or hardware such as GPIO interrupt or data in serial port), module can enter sleep mode. Otherwise will quit sleep mode.

I don't know if it's perfect solution but worked for my setup: 2-channel SMS controlled relay switch. Module is in sleep mode, wakes up with SMS received, switches relay according to SMS content, deletes SMS, goes to sleep again.

Best wishes
Chris

lewisxhe, 2021-01-19:
Issue has been inactive for a long time and will be closed

closed 2021-01-19 by lewisxhe

ishdemon, 2022-05-27:
@toph84 Hey man ..thanks for the tip above..Are you using TinyGSM or other library? I have a similar use-case.. Any chance you can share the snippet or your code?