Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #101: Connect e-paper via SPI

Pigpen1950, 2020-11-18:
I took me two days to find out, so maybe useful to others:
Hardware: Lilygo T-call 1.3 connected to waveshare e-paper HAT and E_paper display.
The EPD2 library of  Jean Marc Zingg worked on "standard" ESP32 but not with the T-call module.
Two issues
1. The Waveshare HAT Rev.2.1 handles resets only in 2ms (see < https://github.com/G6EJD/ESP32-e-Paper-Weather-Display >
2. The Chipselect pin 5 is used for the T-call so needs to be redirected

What worked is to redirect CS to another pin and include a display init with 2ms:    display.init(115200, true, 2, false);
So the VSPI uses Chipselect Pin 5 for T-Call and Pin 33 for the display.

Here the code for a minimal "Hello world"

```
#define ENABLE_GxEPD2_GFX 0
#include <GxEPD2_BW.h> // including both doesn't hurt
#include <GxEPD2_3C.h> // including both doesn't hurt
#include <Fonts/FreeMonoBold9pt7b.h>

#define EPD_BUSY  4 //
#define EPD_RST  14 //
#define EPD_DC   15 //
#define EPD_CS   33 // SPI SS
#define CLK      18 // SPI SCK
#define MOSI     23 // SPI MOSI
#define MISO     19 // SPI MISO - Unused by epaper

GxEPD2_BW<GxEPD2_270, GxEPD2_270::HEIGHT> display(GxEPD2_270(EPD_CS, EPD_DC, EPD_RST, EPD_BUSY)); // GDEW075T7 800x480

void setup()
{
  Serial.begin(115200);
  display.init(115200, true, 2, false);
  SPI.end(); // release standard SPI pins
  SPI.begin(CLK, MISO, MOSI, EPD_CS); // map and init EPD SPI pins
  display.setFullWindow();
  display.firstPage();
  do
  {
    display.fillScreen(GxEPD_WHITE);
    display.setCursor(35, 40);
    display.setFont(&FreeMonoBold9pt7b);
    display.setTextColor(GxEPD_BLACK);
    display.print("Hello world");
  }
  while (display.nextPage());
  delay(100);
  display.powerOff();
  Serial.println("SETUP done");
  esp_sleep_enable_timer_wakeup(1000000);
  esp_deep_sleep_start();
}

void loop() {

};
```

Pigpen1950, 2020-11-18:
CS also works on Pin15 like so
```
#define EPD_BUSY 33 //
#define EPD_RST  25 //
#define EPD_DC   32 //
#define EPD_CS   15 // SPI SS
#define CLK      18 // SPI SCK
#define MOSI     23 // SPI MOSI
#define MISO     19 // SPI MISO - Unused by epaper
```

lewisxhe, 2021-01-19:
Issue has been inactive for a long time and will be closed

closed 2021-01-19 by lewisxhe

sameheddine, 2023-11-17:
Hi team,
i have TTGO T-Call ESP32 SIM800L and i like to connecte my Epaper, with DESPI-C02, 
[image]
wich GPIO i can use ? 
Thank you