Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #103: battery problem

ensaresp, 2020-11-22:
Sorry about the figure. I am using the device in such a connection. battery is not charging and 5v. the device turns off when i cut it. When I boot the device from USB, even if I remove the usb, it continues to work until the battery power cuts off. 5v. The device does not turn off even if I turn the current on and off to the pin.
[image]

lewisxhe, 2021-01-19:
I can't see the picture, I don't understand what you said, can you be more specific?

lewisxhe, 2021-01-19:
I saw the picture. Your battery is connected upside down. The red should be connected to +. Also, you don’t need to connect external capacitors. There are already two capacitors on the board.

ensaresp, 2021-01-23:
> I saw the picture. Your battery is connected upside down. The red should be connected to +. Also, you don’t need to connect external capacitors. There are already two capacitors on the board.

there is no problem with my connections, but when the power supply is turned off, the device turns off. It does not continue from the battery.

lewisxhe, 2021-01-28:
No, your picture shows the polarity is wrong.

lewisxhe, 2021-01-28:
[image]

ensaresp, 2021-02-01:
the picture may be wrong but my connection is correct. but when I give 5v for input the device turns off. so doesn't it have to continue from the battery?