Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #105: TTGO-T-Call 3.3v power supply

eecastillo, 2020-12-03:
Hi, I am new in using the TTGO-T-Call, I have been reading the datasheet and saw that the pin 2 (3v3) can be used for powering the board. However the board doesn't power up, I also tried connecting the battery connector from the backside to 3.3v and the results were the same. This is the tutorial I'm following but I'm not sure if using 3.3v is enough and I wouldn't like to damage my board by using 4.2v without asking.
https://randomnerdtutorials.com/power-esp32-esp8266-solar-panels-battery-level-monitoring/

Thank you.

lewisxhe, 2021-01-19:
Which version are you using?

leenowell, 2021-01-20:
Hi 

I have a v1.3 board and have the same question. Also, what current is required to drive it?  I am prototyping on a breadboard and have a 3.3v / 5v supply so wanted to check whether using this to power via the 3.3v pin is ok? 

Also during development, I also have the usb-c cable connected to the PC to monitor the serial out. Is it correct to have the usb connected whilst also powering via the 3.3v pin?

Thanks for your help

Lee.