Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #107: Modem dies when disconnecting from USB

zcsrf, 2020-12-15:
Hi there,

I have  a T-Call SIM800 v1.4 with a strange issue on the SIM800L.

With a battery soldered directly to the battery terminals (1200mAh / thick wires) + 300uF capacitor on the 5+/GND and the IP5306 boost enable.

If I remove the usb cable while connecting to gprs or with active data connection, the modem simply dies.

No RED led and no response to AT commands. Already tried to toggle the dtr, toggle the pwr pins, nothing works except pressing the reset/en button or programing a new sketch.

This also seems to happen when the modem is roaming between network cells.

Is there a way to make it more stable? or to get the modem to respond again?

zcsrf, 2020-12-15:
One more question,

On the v1.4 schematic, the resistor R114 is NC (IO5), is it really NOT CONNECTED on the final product? 

I tried toggling the MODEM_RST to solve the above and it didn't work.
I also tried toggling the SY8089/DVDD4V4 using EN/IO23 but that also seems not to unlock the modem.

I would like to at least be able to "reset" from such situation, since my implementation can have its my power source disabled at any time and should run on the battery for a couple of hours.

zcsrf, 2020-12-16:
Checked myself and the IO5 is not connected to the reset pin of the SIM800l.

Soldered a jumper and tried to reset the module. it works... the module resets. 

But if I make the Usb disconnection, there must be a glitch and the Reset of the module does nothing to it.

lewisxhe, 2021-01-19:
V1.4 USB is disconnected, using battery power supply, you must press reset to activate power supply