Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #109: Only one Example works

Sasch600xt, 2021-01-08:
Hello,
first i am sorry to open an "Issue"
I think for sure i am the problem :)

Since one week i try to get my LILYGO T-Call SIM 800 to work.
Only the example in here is working for me (downloading the TXT Logo File).
No other example from the tinyGSM library is working for me like "Send SMS" or "Call Test" or "MQTT Client"

So any idea what i could probably doing wrong here ?
Thank you so much for your help !!!
Have a great day
Sascha

lewisxhe, 2021-01-18:
The phone and SMS demos are included in Arduino_TinyGSM.ino. The examples in TinyGSM cannot be used directly in T-Call, you need something like Arduino_TinyGSM.ino
In the same way, initialize him in setup.

Sasch600xt, 2021-01-18:
Thank you so much for your answere :)
could you show me an example how i can get an TinyGSM example to work ?
So where and how to "initialize him in setup" ?
Thank you so much for your help !!!

lewisxhe, 2021-01-19:
You have to refer to the example in the warehouse and initialize the Modem in the setup method
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/0d1a46fd31525ff9336a3353c11951e63760f116/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino#L130

lewisxhe, 2021-01-19:
In the Arduino_TinyGSM example, SMS and call examples are already included. Take a closer look.