Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #110: Is it possible to disable pin 32 DTR function?

ykolerov, 2021-01-18:
Hi, I dont need modem DTR associated with pin 32. Is it possible to disable this function in code or maybe by desoldering something from the board?

lewisxhe, 2021-01-18:
DTR is only used as the IO port for the modem to wake up from sleep, you don't need to use it.

ykolerov, 2021-01-18:
I need to use pin 32 as gpio. When I use it as gpio, modem goes to sleep when I have high level on it. I dont need this functionality. Any way to get rid of it?

lewisxhe, 2021-01-18:
If you want to use GPIO32 and DTR wakeup is not applicable, then you can remove the series resistor. Which board are you using?

ykolerov, 2021-01-18:
I have v1.3 and I also have ordered v1.4. need to disable DTR on both.
Any chance to disable "spectial functions" on other pins (0/4/5/12/33)?

lewisxhe, 2021-01-19:
V1.3 is not connected to DTR, you can use GPIO32 directly. V1.4 you need to remove the resistor to use GPIO32

ykolerov, 2021-01-19:
Thanks for clarification. Which exactly resistor should I remove?

lewisxhe, 2021-01-19:
@ykolerov
 I will reply to you after asking the hardware engineer tomorrow.

ykolerov, 2021-01-19:
Thank you! Please also ask him/her about other pins :-)

lewisxhe, 2021-01-20:
My network is not good, I cannot upload pictures, please leave your email, I will send it to you by email

ykolerov, 2021-01-20:
I sent you an email to your address that is indicated in your github profile. Thanks!

closed 2021-01-22 by ykolerov