Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #112: Battery over-charge / over-discharge protection on either 1.3 or 1.4

dtheodor79, 2021-01-19:
Hi, I have bought both 1.3 and 1.4 boards, which use respectively the IP5306 and AXP192 chips for battery management (among others of course). So I want to make sure that these boards indeed provide the necessary circuitry for connecting non-protected 18650 3.7v batteries in my project. 

The [IP5306 datasheet](http://www.injoinic.com/wwwroot/uploads/files/20200221/0405f23c247a34d3990ae100c8b20a27.pdf) (used in 1.3) states:
o Output over-current, over-voltage short-circuit protection
o Input over-voltage, over-current, battery over-charge, over-drain, over-current protection.

Does "over-drain" mean over-discharge protection for the input battery?

The [AXP192 datasheet](http://www.x-powers.com/en.php/Info/product_detail/article_id/29) (used in 1.4) states:
o Support 4.1V/4.15V/4.2V/4.36V battery
o Charging process control automatically
o Auto adjust the charging current according to the system load

Does "Charging process control automatically" mean that the battery is fully protected against over-charge? What about over-discharge?

Just want to ask others who are already experienced with either of these boards, if indeed they provide the required protection for blocking over-charging / over-discharging a battery.

Thanks,
dtheodor

lewisxhe, 2021-01-19:
Yes, both of them have overcharge and overdischarge protection. You can use them directly without external protection.

dtheodor79, 2021-01-19:
@lewisxhe thanks a lot for the quick update. 

I will close the issue now then.

dtheodor

closed 2021-01-19 by dtheodor79