Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #113: Random GPIO events detected

leenowell, 2021-01-19:
Hi,

I have a project where I need to detect an external event and send an SMS when it happens.  Unfortunately, I am getting a lot of random GPIO events detected.  If I start up the SIM800L I get the issue but if I do not start it, I get no random events and it works fine - i.e. GPIO change only detected when the external event happens.  I am using GPIO 34 to detect the trigger (set in the code as pull down) and GPIO 25 to light up an LED when the ESP detects the trigger.  If I understand the spec correctly, GPIO 34 should be ok (unlike GPIO 32 that I was originally using :) ) so wondering if there is an issue with the spec and 34 is used for something else?

Have also tried pin 35 (both pins in the spec say they are just GPIO) and still the same issue although 34 gives a stream of low states and 35 a stream of high states.

Any thoughts on what could be causing this?  Similarly is there a pin on that side of the board that I can safely use for the trigger detect and lighting the LED?

Thanks in advance for your help

Lee.

lewisxhe, 2021-01-20:
Which version are you using? GPIO36 and 34 can be used, they are not connected anywhere, you only need to add an external pull-up resistor, or a pull-down resistor, according to your circuit. If you use the power management of AXP192, then 35 has been connected to the interrupt of 192.

leenowell, 2021-01-20:
Hi

Thanks very much for the reply. I have v1.3 of the board and am using the ESP_IDF v4.1.  I am not explicitly using any power management features.  In my code, I am using the internal pull-down resistor of the ESP32 so thought this would be enough?  Are you saying I also need to have an external pulldown resistor?  If so, what value as I think the internal one is 45k? 

To understand which of the GPIOs cna be used, how should we read the pin-out diagram (https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/image/SIM800L_IP5306.jpg)?  For those that have a GPIO number next to them it would imply that it is safe to use the as a regular GPIO pin unless using them for something else e.g. GPIO32 unless I am using  a touch device which would then need to be connected to it?  On the pin out diagram in git, it doesn't have GPIO36?

thanks again for your help

Lee.

leenowell, 2021-01-20:
Hi All,

I have tried GPIO 34 with a 8k external pull down resistor and with the internal pull down set and have the same issue.  This thread seems to suggest that GPIO 34 is input only and does not have any pull up/ down resistors but can't find any confirmation of this.  Irrespective, shouldn't the 8k resolve the issue?

@lewisxhe Is there anything else I can try? 

Thanks

Lee.

leenowell, 2021-01-21:
@lewisxhe Thanks for your reply.  As mentioned, I used an 8k pull down resistor (GPIO34 to gnd) and still had the same issue.  Is there a spec which shows which pins can and can not be used?  The pin out diagram (https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/image/SIM800L_IP5306.jpg) implies that 4,5,23,26,27 are the only ones that are "reserved" for the board and I could use any of the others.  However as you mention above 35 could be used by the board and I also saw somewhere that 25 is also used.  From the pinout diagram it implies that any pin marked with GPIO can be used for both input and output?

Could this be a power supply issue (only thing I can now thing of) as per issue 105?  https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/105

Thanks

Lee.

lewisxhe, 2021-01-22:
No, it has nothing to do with the power supply. The board using AXP192 power management uses GPIO35 as interrupt input.

Only below 33 can be configured as input and output, and above 33 can only be used as input, please check the manual.

leenowell, 2021-01-22:
Thanks @lewisxhe As mentioned above, I used pin 34 with an 8k external pull down resistor and still got the random GPIO events.  I believe I have got to the bottom of the issue and think it is a power issue.  I was getting "brownout" issues when connecting the board via USB to my laptop.  So, I decided to use a breadboard power supply (3.3v) to power the board connecting it to the +3.3v and gnd pins.  In addition, had it connected to the laptop via USB to flash it and debug via serial monitor.  Switching this back to only power via USB has solved the GPIO problem but I am back to having brown out issues.

So.... this would imply that the +3.3v pin cannot be used to power the board and maybe is to source 3.3v for the external circuit to use?  If this is the case, clarifying this (and the role of the +5v pin) would be very helpful on the pin out diagram?

My project is very simple (a switch on one pin and led on another) so what is the best way to power the board as via USB only is causing "brownouts"

Thanks

Lee.

leenowell, 2021-01-28:
@lewisxhe  Any thoughts on the power supply issue?

lewisxhe, 2021-01-29:
Sorry, I forgot to reply, the 3v3 board can add pull-up resistors externally, 3v3 supports adding a small amount of load, there should not be a power failure problem, can you give me a schematic diagram and let me see how you connect it  .

leenowell, 2021-01-30:
Hi @lewisxhe 

I have drawn up the circuit here https://crcit.net/c/9d45f50f79cf4e4ca6fe939881454deb. The board is powered via USB from my PC.  The input circuit into GPIO12 is an opto coupler (the tool didn't have one in it's library) and GPIO12 has been configured as input with a pulldown resistor.

Besides this pin out diagram https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/doc/SIM800L_IP5306.MD (this is the board I have - v1.3 20190610) is there any other documentation on the board. E.g. whether the 3.3v/ 5v and inputs/ output; which pins are used internally and we should not use (e.g. it doesn't indicate that pin 35 is also used by the power management chip) etc.

Thanks for your help

Lee.

lewisxhe, 2021-01-30:
There seems to be no problem with the circuit, but you use GPIO12. Please note that if you trigger an external switch when ESP32 starts, GPIO12 should be high at this time, then ESP32 will not start smoothly, because GPIO12 is used as a selection when starting Flash voltage selection function, I suggest you replace a pin.

PS: Your drawing is a bit similar to my recent project, you can refer to: 

[image]

leenowell, 2021-01-30:
Hi @lewisxhe Thanks for getting back to me and for sending the diagram. Will try another pin but to be honest without a definitive list of which of the break out pins are safe to use (i.e. not used by the board at all) it is a bit hit and miss.  Assume this isn't causing the brownout though?

thanks again

Lee.

lewisxhe, 2021-01-30:
Except for special function IO (IO2, IO0, IO12), all others can be used, you can check the manual of esp32.

Of course the SIM800 is not available. You can see this from the code. 
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_AT_Debug/utilities.h

leenowell, 2021-01-30:
Thanks - the header file definitely helps.  Looks like me using GPIO22 is a little dodgy too :)  Do you power your board from the USB port or do you power via the 5v pin (I believe 3.3v pin is output only on v1.3 board and 5v is input?).  I also read somewhere that the board needs a power supply of 2A which seems very high and not sure standard USB ports can serve up that much?  I know it probably doesn't help but if I don't power up the SIM800 I don't get any brown outs so it seems that the power demands of SIM800 tips it over the edge!

lewisxhe, 2021-01-30:
You can plug in the battery while plugging in the USB to keep the power supply stable.

leenowell, 2021-01-30:
unfortunately I have the v1.3 which doesn't have a battery socket on the back :( I have one of these https://www.ebay.co.uk/itm/202792865563 which may have originally come with my T-Call but don't have the battery to go in it.  Assume this would need to be connected to the 5v pin? 

I have an old 5v charger off some device can I connect that to the 5v pin and also connect to PC via USB for serial monitor?

lewisxhe, 2021-02-04:
This seems impossible, because 5V is connected to the voltage output terminal of IP5306. If you do this, it will not work properly, and it may be damaged.

closed 2021-10-19 by LilyGO