Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #114: having some problem with the selecting the board

5h15h1r, 2021-01-21:
[image]

guide me please

bhupiister, 2021-01-21:
try selecting wrover dev board. Which board are you selecting and which version of esp32 Tcall you have ?

lewisxhe, 2021-01-21:
Just choose ESP32 Dev Module.  For your version, you can find the date on the board and uncomment that line after finding the date corresponding to the program.

5h15h1r, 2021-01-22:
> try selecting wrover dev board. Which board are you selecting and which version of esp32 Tcall you have ?
SIM800C_AXP192_VERSION_20200609
tried wrover dev board, isn't working for me 
the LED of  my T-call is not blinking now ,idk why