Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #116: Any chance for an example with MQTT and TLS/SSL?

kedzior-io, 2021-01-27:
It would be great if you guys could provide an example for MQTT with TLS/SSL configuration? 
Most of the IOT cloud providers require it today.

Here is an example of MQTT I'm talking about: 
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_MQTT/Arduino_MQTT.ino

Here specifically talking about connecting it to Azure IOT Hub:
https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#using-the-mqtt-protocol-directly-as-a-device

and required TLS/SSL configuration:
https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#tlsssl-configuration

Thanks!

lewisxhe, 2021-01-28:
Currently, the built-in firmware of 800L does not support SSL and MQTT. I will look for SIMCom to provide technical support and ask them to provide me with a new version of the firmware. I will add it to the example when I finish the test.

kedzior-io, 2021-01-28:
> Currently, the built-in firmware of 800L does not support SSL and MQTT. I will look for SIMCom to provide technical support and ask them to provide me with a new version of the firmware. I will add it to the example when I finish the test.

Awesome! Thank you!

lewisxhe, 2021-01-30:
I am in China and I cannot access azure. I will try to connect to another IOT platform for testing. Do you have a recommended IOT platform? 

@kedzior-io

lewisxhe, 2021-01-30:
I did not find the English document of 800L SSL, you can view the Chinese one. Hope it helps you
https://cn.simcom.com/product/SIM800L.html

closed 2021-08-25 by kedzior-io