Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #118: Dead board

leenowell, 2021-01-31:
Hi,

Having flashed my board loads of times, my board has decided it doesn't wan't to flash any more and I see nothing in the serial monitor.  I have tried the suggestions here https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/31 i.e. powering it up with 0 and gnd pins connected and pressing the reset button a few times whilst connecting and no luck.  I have also noticed that sometimes when I start it a blue LED below the ESP32 chip is continually on - never noticed this being on at all before.  Board is v1.3 20190610.

Only thing I have done differently is that I connected some capacitors between 5v and gnd pins in an attempt to solve some brownout issues.

Any help appreciated.

Thanks

Lee.

lewisxhe, 2021-02-03:
First of all, you need to make sure that the communication between the board and the UART bridge is normal, then short-circuit GPIO0 & GND, then press the reset button, and then click upload in the Arduino IDE. Don’t disconnect GPIO0 and GND before the upload is complete, this is just Prevent missed download mode.

leenowell, 2021-02-03:
Hi @lewisxhe  

Thanks very much for getting back to me.  How would I test if the board UART is functioning?  I have tried following the steps you mention(using Eclipse and idf.py (previously I had missed the press reset button) and still no luck :(. The ESP32 is gettng warm so is doing something just nothing on the monitor.  What does the blue LED indicate? Is this it going into upload mode as when I tested it the LED was not lit.

thanks again

Lee.

lewisxhe, 2021-02-04:
How to judge whether the communication is normal, you can open the serial port tool, connect the board, and then press the reset button to check whether there is a startup message output, if the output is normal, then the communication is normal.
At this time, you can short-circuit GPIO0 to GND, and then press the reset button again. At this time, the serial port will output a message waiting for download.