Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #119: Powering LilyGo TCall sim800 without usb type c

daniel571, 2021-02-01:
Hi,

I have tried to connect a lithion ion battery of 3.7v to the white connector with the wires attached with the product . The battery had 4v open circuit. However , I did not see that the board was start up. The leds stuck at off.
I tried to press reset button and also did not start power.
When usb type c connect it works fine.
How to start it up with battery connected to the battery header ?

Any creative solution will be respected

Thanks

daniel571, 2021-02-02:
I have been thinking to connect the battery with pn junction to the 5v pin. This will wake up the esp32. Then the esp will enable the boost for battery manager with ic command. This will create 5v . The pn juntion will protect the battery from higher voltage

lewisxhe, 2021-02-03:
There is a MOS tube in series between the IP5306 and the battery connector. It is not directly connected. You can check the [Schematic](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/schematic /LilyGo-SIM800L-IP5306-20200811.pdf)

Under normal circumstances, connect the battery, you need to press the reset button to wake up.

When the battery is connected, you can measure whether there is battery voltage on pin 6 of IP5306

daniel571, 2021-02-03:
I plug a dc source of 4v to simulate lithioum ion battery to the battery header and press the reset button. It consume for short time 100ma but the module did not wake up.

daniel571, 2021-02-03:
Unfortunately, when I press the reset button the 5V signal is rising up to 4V and then drop back to 2.2V and stay there.

LilyGO, 2021-10-19:
Hello, have you solved your issuse now?

proximus323, 2021-12-13:
Power boost!!!

I spend cople of hours to get this work. 

bool setupPMU()
{
  bool en = true;
  Wire.begin(I2C_SDA, I2C_SCL);
  Wire.beginTransmission(IP5306_ADDR);
  Wire.write(IP5306_REG_SYS_CTL0);
  if (en)
  {
    Wire.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  }
  else
  {
    Wire.write(0x35); // 0x37 is default reg value
  }
  return Wire.endTransmission() == 0;
}