Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #122: sleep current

ninnux, 2021-02-14:
I try to use esp-idf to put esp32 in deepsleep and my board (SIM800L_IP5306) from battery plug absorbs 43mA.
Some one could share some code example to put down all parts (modem SIM800L, etc) making this board  consuming 0.0011A only?

aicherm, 2021-07-23:
I get the same sleep current (47mA @4V) with my board SIM800L_IP5306:

```
 modem.poweroff();
 esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
 esp_deep_sleep_start();
```

I would really appreciate if someone can show some code examples or give a hint what to turn off additionally!

MatejTT, 2022-01-29:
I'm trying to push down current as mush as possible in current project. Was someone able to find out? Thanks!