Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #127: Are there UART availables on TTGO T-CALL SIM800L board ?

clabnet, 2021-03-23:
Hi,

I need to use integrated modem of TTGO T-CALL SIM800L board and another serial device.

Are there available UART(s) for connecting via serial to this board ? If yes, what are the PIN(s) ?

Thank you, regards

LilyGO, 2021-10-15:
The serial port is also available, and you need to define it.
 Serial2.begin(115200, SERIAL_8N1, RX_PIN, TX_PIN);

lmorejon01, 2024-06-17:
Hi all, please I need some help, I am testing ESP32 T Call SIM800 L with 2 PZEM 004T ( they have serial port RS232 TTL level each one), the electrical data from the PZEM I want to send via GSM network, the problem is I can´t get success connecting 2 PZEM with serial port...only operate 1 PZEM, I define one serial port in pins RX like GPIO 14 and TX like GPIO 12, Can I define other serial port by software or hardware in others GPIO? thanks in advance
Best Kind Regards from Guayaquil Ecuador
Lenin Morejon