Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #129: I bought this because it said it was compatible with Micropython.  I will have to return it now.

Tylersuard, 2021-04-03:
Hello.  I am looking to make an ESP32 communicate with a 2G network.  I thought I would buy the T-Call Sim800 because it had the sim800l integrated so I don't have to make any extra connections.
The software you have for MicroPython is completely unusable.  The firmware you recommend hasn't been updated in 3 years.
The product itself is probably great, but I have to return it because the software isn't usable at this point.

ClemensGruber, 2021-04-05:
You may have a look at https://github.com/hiveeyes/terkin-datalogger a MicroPython data logging framework that runs on the T-Call also or this lib https://github.com/pythings/Drivers

Tylersuard, 2021-04-05:
@ClemensGruber That was very, very helpful.  Thank you!

McMacaron, 2021-04-21:
Can you write what was helpful?

Tylersuard, 2021-04-21:
@McMacaron 1. Download the most recent micropython drivers and install them using Thony
2.  Upload the .py files to the board from here: https://github.com/pythings/Drivers

Kamalesh1907, 2021-04-26:
Hi can u please help me on how to install the phyton?? Is it a must to download the python??

McMacaron, 2021-04-26:
Hello Kamalesh,
are you going to program this device in Python? Or you will use classic Arduino code "C++"? If you need Python just Install VS code find Python i the store and install it. There is plenty tutorials on you tube.

Tylersuard, 2021-04-26:
@Kamalesh1907 Just search for a video online.  Here is one: https://www.youtube.com/watch?v=lvmNLuHj25o

Kamalesh1907, 2021-04-28:
Hi im using the classics arduino ide so do I need to install the phyton and
when uploading some code it says selected serial port failed to execute
script esptool does not exist on your board

closed 2021-10-20 by LilyGO