Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #131: Update Firmware

WRedux, 2021-04-05:
Hello,

I tried to update my Firmware B05 to B06.
Won't work, so some maybe stupid Question:
Connection? Normal via USB-C?

Did this. Followed the Update-Procedere as descríbed.

But when I start Download, the LEDs of my Board go out - 
nothing's happen.

Insufficent Current?
Although I'm using a Hub with external Power Supply.

Tried several Times, with the Result of leaving the Board in a Boot-Loop...

Did I miss something?

Sorry...

Greetings,

micdj, 2021-04-26:
Hello, me too!! The same problem.

any ideas?

WRedux, 2021-05-02:
Bought another Hub now with 2,5Amps,
Needed anyway... but didnt test it so far...

buttim, 2021-11-02:
I had the very same problem. In my case I could solve it modifying the UpdateFirmware Arduino sketch.
I used the latest SIM800L firmware (1308B09SIM800L16) since the version on this github project gave me errors during the update.
Here are my modifications:

1. I commented the setup modem call:
```
// Some start operations
//setupModem();
```
2. added this declaration in the loop function:
`static bool init=false;`
3. modified the _second_ while in the main loop like this:
```
    while (SerialMon.available()) {
      if (!init) {
        setupModem();
        init=true;
      }
        SerialAT.write(SerialMon.read());
    }
```

marmou-fritzing, 2021-12-31:
I have made my own utility which works for me
[T-Call_SIM800_firmware_upgrade.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7796465/T-Call_SIM800_firmware_upgrade.zip)
.

marmou-fritzing, 2022-01-02:
Because I'm solving some problems with feedback at ATD command, I was looking for newest firmware. Here saved firmware for SIM800L has begin 1418B, what is according to SIM800C firmware codes (from simcom.ee). SIM800L has begin 1308B. I tried upload 1308B firmware, downloaded from simcom to SIM800L, but is incompatible. How is it please ?

byte12345, 2022-06-08:
I can not update SIM800L FW. I always receive S_BROM_DOWNLOAD_DA_FAIL ( 2004 ) error. The product is T Call 1.3 with SIM800L and IP5306 power chip. Changed the PC, the USB cable .. no result.

marmou-fritzing, 2022-06-08:
And when it happens ?

2r-lgk, 2022-07-22:
Hello,

When I download the 1418Bsim800L firmware version it works fine, but when I'm using the latest one (1308B09SIM800L1) the flash tool finish the download and throws me this error. 

(FLASHTOOL ERROR: S_DL_MAUI_FLASH_ID_NOT_MATCHED_WITH_TARGET (5095)). 

Did you had the same problem?

thanks

somehibs, 2023-02-06:
I can't flash SIM800L_IP5306_VERSION_20190610 at all. I tried the firmware patcher from this thread, and I get

"Communication with GSM module confirmed.
Close terminal window and let's upgrade the SIM800 firmware."

but the "Simcom Series download Tools Customer v1.22" application freezes at "0% (Waiting)".

marmou-fritzing, 2023-02-06:
I have for you renewed manual.
Please see the bold rows.

// --- Transaction sketch ---
// !!! At first please uncomment the corresponding model below (Number should be printed on bottom side of TTGO board)
// Select corresponding model in IDE and select COMPort.
// Open UART terminal.
// Press button on board to be an answer from device "...waiting for download" or try upload sketch directly.
// Upload this sketch into board.
// After will be confirmed communication with device, close the terminal window to be port undone and start upgrade by Simcom download tool.

// --- SimCom download tool ---
// **Recomended version of Simcom download tool is 1.06**
// **Attention to the path, where are the files on disk stored. In case of other, than ASCII paths or long paths could be a problem open application or files for uploading.**
// **Run the Download tool with Admin rights.**
// Select SIM800 variant, UART type and UART port.
// **Attention to the versions of firmware. For example firmware for SIM800L device the firmware file could begin (in according to SIMCOM ftp) by "13", but compatible is "14".** (Why ??)
// Select corresponding firmware.
// Start upload.

marmou-fritzing, 2023-02-06:
Fresh update 1 minute ago.
[image]
Whe it will not works, disconnect device, wait 5 seconds and then connect device again. Wait so 10 seconds and start upload.
Try it please.

somehibs, 2023-02-07:
Thank you, I appreciate your effort.

My board is from 2019 (SIM800L_IP5306_VERSION_20190610), marked "BLYNK V1.3 20190610", which might be the reason it's not working.

I have tried UpdateFirmware and T-CALL_SIM800_firmware_upgrade attached to this ticket, confirming the modem functions properly with the second sketch
I have tried both SIMCOM flash tools. Both flash tools and sketches exhibit the same behaviour
I have placed the firmware in the root of a drive, like yours

Clicking start download makes the blue serial LED indicator turn off and the upload does not progress. New SIMCOM tool freezes on 'waiting' and old SIMCOM tool counts up forever. Got to 1400 sec once.

Squirrelbd, 2024-08-10:
Hello,

It sounds like you're encountering some common issues during the firmware update process. The symptoms you're describing—LEDs going out and the board entering a boot loop—could indeed be related to insufficient current, especially if the board is not receiving stable power during the update.

Here are a few steps you can try:

    Check Power Supply: Even though you're using a hub with an external power supply, it's worth double-checking the current rating. Make sure it can supply enough current, as the SIM800 modules can be quite power-hungry, especially during updates.

    Direct Connection: Instead of going through a USB hub, try connecting the board directly to your computer's USB port. Sometimes, hubs can introduce issues, especially if they're not delivering a stable power supply.

    Use a Reliable USB-to-TTL Converter: If you're updating via serial, ensure you're using a reliable USB-to-TTL converter. Some converters may not provide adequate voltage levels or may cause communication issues.

    Follow the Correct Procedure: Ensure you're following the correct update procedure, including putting the module in the correct mode for firmware update. You might want to double-check the steps and requirements.

For a detailed step-by-step guide on updating the SIM800 firmware, you can check out this [guide on SIM800 GSM Module Firmware Update](https://www.programmingboss.com/2024/08/update-sim800-gsm-module-firmware-using-SIMCom-firmware-download-tool.html). It covers everything from preparation to troubleshooting common issues like the one you're experiencing.

If you're still facing issues after trying these steps, there might be other factors at play, and more detailed diagnostics could be necessary.

Hope this helps!