Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #132: ESP32 LilyGo GSM module OTA possibilities

krupis, 2021-04-08:
Hello. I am using this module to learn how to use GSM modules. I have been experimenting with the example codes and I have learnt the basic working principles. The big project that I am going to be working on next will involve the GSM and the GPS module.

 I am very interested in how to perform a OTA over the GSM since the there will be very limited WIFI availability where my device will be used. At some point, I may want to update the firmware, fix bugs, so I the probably the easiest way to be to use the GSM and do the OTA over the GSM. 

I have noticed the update_firmware under the example codes but there is nothing there. Perhaps anyone has made any progress and could share their knowledge and ideas with me? I appreciate any help! Thanks in advance

thomsenle, 2021-05-01:
see tinygsm examples on how to open a tcp connection and use it, and see esp32 https ota examples 
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples/HTTPS_OTA_Update

thomsenle, 2021-05-01:
see also here https://github.com/vshymanskyy/TinyGSM/issues/152

gunter72, 2021-05-09:
Hi, I have the same problem.
How did you solve?

emigbur, 2021-05-15:
I did a skectch and work fine, is any are interested I can share ti, but I dont know how to make the server to store the bin file.
I did the sketch using the TTGO CALL SIM800L github GSM tiny OTA example.
If someone has and server example I will apreciate it, my idea is to store the bin file on my web server.

krupis, 2021-05-15:
Hello. Here is the code to perform OTA over the GSM:

```

#include <Update.h>
#define TINY_GSM_MODEM_SIM800
// Increase RX buffer
#define TINY_GSM_RX_BUFFER 1030

const char apn[] = "omnitel";
const char user[] = "omni";
const char pass[] = "omni";

#define MODEM_RST             5
#define MODEM_PWRKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define LED_GPIO             13
#define LED_ON               HIGH
#define LED_OFF              LOW

#define SerialAT  Serial1

#include <TinyGsmClient.h>
#include <CRC32.h>

#include "FS.h"
#include "SPIFFS.h"

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

TinyGsmClient client(modem);

const char server[] = "myesptestserver.ddns.net";
const int port = 80;
const char resource[] = "/esp.bin"; //here de bin file

uint32_t knownCRC32 = 0x6f50d767;
uint32_t knownFileSize = 1024; // In case server does not send it

void setup()
{
    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
    // Set console baud rate
    Serial.begin(115200);

    setupModem();
    delay(10);

    if (!SPIFFS.begin(true))
    {
        Serial.println("SPIFFS Mount Failed");
        return;
    }
    SPIFFS.format();
    listDir(SPIFFS, "/", 0);

    // Set GSM module baud rate
    
    delay(3000);

    // Restart takes quite some time
    // To skip it, call init() instead of restart()
[... 382 lines cut ...]

Make sure to change accordingle the following lines of code:
```
const char apn[] = "omnitel";
const char user[] = "omni";
const char pass[] = "omni";
```
and 
```
const char server[] = "myesptestserver.ddns.net";
const int port = 80;
const char resource[] = "/esp.bin"; //here de bin file
```

For storing the bin files, you have multiple choises. 

1. Use Raspberry pi as a server ( I have created a simple apache server where I uploaded my binary file, and then I can access that server through the following sketch.

However, for testing it maybe be easier to setup Xampp on windows machine just to see whether you can get it to work.
2. Create an apache or simmilar server on your windows machine ( probably the simplest method would be to use Xampp as it does everything for you. In the xampp folder you will have htdocs folder where you can but your binary. and you can access that binary by using my sketch or your own sketch 
```
const char server[] = "myesptestserver.ddns.net"; // ip of server or web address
const int port = 80;
const char resource[] = "/esp.bin"; // put your binary file name
```

eueduardo3, 2021-05-16:
> Hello. Here is the code to perform OTA over the GSM:
> 
> ```
> 
> #include <Update.h>
> #define TINY_GSM_MODEM_SIM800
> // Increase RX buffer
> #define TINY_GSM_RX_BUFFER 1030
> 
> const char apn[] = "omnitel";
> const char user[] = "omni";
> const char pass[] = "omni";
> 
> 
> 
> #define MODEM_RST             5
> #define MODEM_PWRKEY          4
> #define MODEM_POWER_ON       23
> #define MODEM_TX             27
> #define MODEM_RX             26
> #define LED_GPIO             13
> #define LED_ON               HIGH
> #define LED_OFF              LOW
> 
> 
> #define SerialAT  Serial1
> 
> #include <TinyGsmClient.h>
> #include <CRC32.h>
> 
> #include "FS.h"
> #include "SPIFFS.h"
> 
> #ifdef DUMP_AT_COMMANDS
> #include <StreamDebugger.h>
> StreamDebugger debugger(SerialAT, Serial);
> TinyGsm modem(debugger);
> #else
> TinyGsm modem(SerialAT);
> #endif
> 
> 
> TinyGsmClient client(modem);
> 
> const char server[] = "myesptestserver.ddns.net";
> const int port = 80;
> const char resource[] = "/esp.bin"; //here de bin file
> 
> uint32_t knownCRC32 = 0x6f50d767;
> uint32_t knownFileSize = 1024; // In case server does not send it
> 
> void setup()
> {
>     SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
>     // Set console baud rate
>     Serial.begin(115200);
> 
>     setupModem();
>     delay(10);
> 
>     if (!SPIFFS.begin(true))
>     {
>         Serial.println("SPIFFS Mount Failed");
>         return;
>     }
>     SPIFFS.format();
>     listDir(SPIFFS, "/", 0);
> 
>     // Set GSM module baud rate
>     
[... 589 lines cut ...]
      } else {
        Serial.print("error in downloading version file:");
        Serial.println(httpCode);
      }
      https.end();
    }
    delete client;
  }
      
  if (httpCode == HTTP_CODE_OK) // if version received
  {
    payload.trim();
    if (payload.equals(FirmwareVer)) {
      Serial.printf("\nDevice already on latest firmware version:%s\n", FirmwareVer);
      return 0;
    } 
    else 
    {
      Serial.println(payload);
      Serial.println("New firmware detected");
      return 1;
    }
  } 
  return 0;  
}

krupis, 2021-05-16:
Hello. Have you managed to get examples from Arduino IDE working? They have
an example sketch that performs an OTA over the http?

From the first glance, your code looks fine. What error are you getting?

eueduardo3, 2021-05-16:
> Hello. Have you managed to get examples from Arduino IDE working? They have an example sketch that performs an OTA over the http? From the first glance, your code looks fine. What error are you getting?
> […](#)

I tried to use your script to try to OTA a file hosted on GitHub, but I couldn't. In this example that I posted, it is okay to OTA a file through github through Wifi, but I was unable to do that using a gsm modem through the TinyGsmClient library

eueduardo3, 2021-05-16:
I tried to use your script to try to OTA a file hosted on GitHub, but I couldn't. In this example that I posted, it is okay to OTA a file through github through Wifi, but I was unable to do that using a gsm modem through the TinyGsmClient library

krupis, 2021-05-16:
That is really weird. I have created an apache server using xampp and can
do an OTA from there. What error are you getting? Are you using esp32 with
spiffs fs partition? Do you know where your code gets stuck

eueduardo3, 2021-05-20:
Now it worked!
Thank you very much for making your code available and helping me. I changed the server and it worked. I was using 000webhost. And it was certainly my mistake, I was not sure how to correctly make the file available for download. Switching from server to another site / server with free hosting worked. Thank you very much for your attention in trying to help me. I wish success for your projects.

eueduardo3, 2021-05-20:
> That is really weird. I have created an apache server using xampp and can do an OTA from there. What error are you getting? Are you using esp32 with spiffs fs partition? Do you know where your code gets stuck
> […](#)

Now it worked!
Thank you very much for making your code available and helping me. I changed the server and it worked. I was using 000webhost. And it was certainly my mistake, I was not sure how to correctly make the file available for download. Switching from server to another site / server with free hosting worked. Thank you very much for your attention in trying to help me. I wish success for your projects.

gunter72, 2021-06-24:
> I did a skectch and work fine, is any are interested I can share ti, but I dont know how to make the server to store the bin file.
> I did the sketch using the TTGO CALL SIM800L github GSM tiny OTA [...]

Hi, could you share your code? I'm having lots of issues trying any other code

Thank you

krupis, 2021-06-24:
> Hi, could you share your code? I'm having lots of issues trying any other code
>
> Thank you

Hey. Have you tried the OTA over GSM code that I have posted above? It would be a good start if you can get that to work, then you can modify it as you wish. Make sure you have some sort of webserver available where you going to store your binary file for the OTA. Myself, I have downloaded XAMPP on windows and store my binary files there, but there are many options available.

gunter72, 2021-06-24:
Hi, I tried your code, but it always reaches 90% and then it gives me an error.  I have a private webserver, but I also tried altervista, I always have the same problem.

krupis, 2021-06-24:
Could this be an issue due to the size of a binary you are trying to flash?
OTA has some limitations especially if its a huge binary. Have you tried to
play with changing the size of OTA ? Also, you can eliminate this by trying
to flash some led blink or small binary program.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gunter72, 2021-06-24:
Yes, I tried with small 100k files but I have the same error.  My bin is 660k

krupis, 2021-06-24:
> Yes, I tried with small 100k files but I have the same error. My bin is 660k

I see. Thats unfortunate.. What GSM are you using? Have you tried to just read the whole binary file and not do the OTA and see if you can read the whole binary?
Instead of updateFromFS try to read your binary
```
    readFile(SPIFFS, "/update.bin");

    //updateFromFS(); // comment this out
```

gunter72, 2021-06-28:
Hi krupis, thank you very much for your support. I found the problem. It was a mistake in my code.
The update took place while the mqtt communication was still active.
I have stopped all active clients first, then I start the update process and everything works.
Thanks so much!

krupis, 2021-06-28:
Glad you found the issue.

Mohamed-ali1998, 2021-09-04:
> Hello. Here is the code to perform OTA over the GSM:
>
> ```
>
> #include <Update.h>
> #define TINY_GSM_MODEM_SIM800
> // Increase RX buffer
> #define TINY_GSM_RX_BUFFER 1030
>
> const char apn[] = "omnitel"; [...]

@krupis Hi krupis, I'm ok with esp32 coding side. But I have idea to configure xamp server and where to the .bin file in the server. How can I do this? Can you share any reference for this?
Thanks in advance.

krupis, 2021-09-04:
@Mohamed-ali1998  Hey mate. The binary should be stored inside htdocs folder where you have installed XAMP. There should be no additionall setup required apart from starting the XAMPP services. Also, keep in mind that in order to perfrom GSM OTA, you will need to port forward the IP address. I have used some free domain service. The reason is because the GSM is not connected to your home network and will not be able to reach your local computer IP address.

Mohamed-ali1998, 2021-09-21:
@krupis Hi krupis. I used our own server. That's working fine. Thank you so much.

TawalMc, 2021-10-15:
> @TawalMc Hello. I dont think I can help you with this issue but @gunter72 had a simillar problem.
> 
> Before you do anything else, I would try 2 things
> 
> 1. Try with small binary
> 2. Isolate the rest of your code just to confirm if GSM OTA part is working. If it does, then start adding your other code bit by bit and see which part cause the issues.

Thanks. I'll try them. Good

KASSIMSAMJI, 2021-11-03:
> Thanks. I'll try them. Good

Hello, i am getting same error#9, any luck ?

ilker07, 2021-12-28:
f

> Hello, i am getting same error#9, any luck ?

I have same error sometimes.Sometime it works.Could you find any solution?

KASSIMSAMJI, 2021-12-28:
> f
> 
> > > > @TawalMc Hello. I dont think I can help you with this issue but @gunter72 had a simillar problem.
> > > > Before you do anything else, I would try 2 things
> > > > 
> > > > 1. Try with small binary
> > > > 2. Isolate the rest of your code just to confirm if GSM OTA part is working. If it does, then start adding your other code bit by bit and see which part cause the issues.
> > > 
> > > 
> > > Thanks. I'll try them. Good
> > 
> > 
> > Hello, i am getting same error#9, any luck ?
> 
> I get same error sometimes.Sometime it works.Could you find any solution?

in my case the success rate was so high (2 fails out of 20) few hours after midnight, so I deduced  It is due to network being hogged during the day

ilker07, 2021-12-28:
> in my case the success rate was so high (2 fails out of 20) few hours after midnight, so I deduced It is due to network being hogged during the day

this morning I upload the code and it updated but after that I created new bin file and it gave me error.The error is  E (342241) esp_image: invalid segment length 0x2d832b6.

ilker07, 2021-12-28:
> in my case the success rate was so high (2 fails out of 20) few hours after midnight, so I deduced It is due to network being hogged during the day

[image]

Should I change these settings?

ilker07, 2022-02-11:
> > this morning I upload the code and it updated but after that I created new bin file and it gave me error.The error is E (342241) esp_image: invalid segment length 0x2d832b6.
> 
> He guys I am facing the same issue. Were you able to resolve it? @ilker07 @gunter72 @krupis

I think 2 thing cause this.1.Power Supply 2.Network.When I use 3.7 v battery ,mostly it works ,sometimes it fails probably due to network.When I don ' t use 3.7v battery and I power the board through USB,it fails definitely.

zark-zeugan, 2022-02-11:
> > > this morning I upload the code and it updated but after that I created new bin file and it gave me error.The error is E (342241) esp_image: invalid segment length 0x2d832b6.
> > 
> > 
> > He guys I am facing the same issue. Were you able to resolve it? @ilker07 @gunter72 @krupis
> 
> I think 2 things cause this.1.Power Supply 2.Network.When I use 3.7 v battery ,mostly it works ,sometimes it fails probably due to network.When I don ' t use 3.7v battery and I power the board through USB,it fails definitely.

Okay. This might be a silly question, but then how do you see the output without the USB connection?

krupis, 2022-02-11:
You cant. You can power the device via the external battery but also connect the USB to check the logs

zark-zeugan, 2022-02-11:
> You cant. You can power the device via the external battery but also connect the USB to check the logs

@krupis @ilker07. But how is network or battery or network related to `Update.end()` not working? 
If I see we have already disconnected the modem, and the file has already been written on the device. 
And while performing `Update.end()` is when things go sideways. 
Also, is your device able to update for higher file sizes? Are you able to perform OTA Updates? 
Could you help me solve this error?
```
Content-Length: 267648
Actually read:  267649
Calc. CRC32:    0x0
Known CRC32:    0x6F50D767
Duration:       94.49s
starting Update after 3 seconds  
0...1...2...start of updating
writings : 267520 successfully
E (143352) esp_image: Checksum failed. Calculated 0xfa read 0xff
Error occured #: 9
```

KASSIMSAMJI, 2022-02-15:
Hello, so I finally had the SIM800L and the ESP32 SOC on the neat PCB, with an LM2596S-adj giving out 3.6V

I never had issues anymore with the OTA, out of 20 trials, no failure, 

looks like It was a power issue to the SIM800L

krupis, 2022-02-15:
Most problems can be solved by by providing a good voltage for the modem. Even the 3.6V is considered to be the absolute minimum for the modem. The recommended is around 4.0V.

zark-zeugan, 2022-02-15:
> Most problems can be solved by by providing a good voltage for the modem. Even the 3.6V is considered to be the absolute minimum for the modem. The recommended is around 4.0V.  

@krupis  Okay. Just one last question do we have to write something in the code to let the device know which power source to use? Like use the USB one or the battery one that is attached?

krupis, 2022-02-15:
You definately dont need to do that in the code. I am not much of a hardware guy so cant give you correct answer. I assume this depends on how the hardware is designed... I assume if you have USB plugged in but you also supply an external voltage higher than 3.3V than it should be fine.

zark-zeugan, 2022-02-16:
> Hello, so I finally had the SIM800L and the ESP32 SoC on the neat PCB, with an LM2596S-adj giving out 3.6V
> 
> I never had issues anymore with the OTA, out of 20 trials, no failure,
> 
> looks like It was a power issue to the SIM800L

@KASSIMSAMJI 
Where are you hosting the binary files?

KASSIMSAMJI, 2022-02-16:
> > Hello, so I finally had the SIM800L and the ESP32 SoC on the neat PCB, with an LM2596S-adj giving out 3.6V
> > I never had issues anymore with the OTA, out of 20 trials, no failure,
> > looks like It was a power issue to the SIM800L
> 
> @KASSIMSAMJI Where are you hosting the binary files?

I have a shared hosting server for that

krupis, 2022-03-02:
> Hello everyone, I'm using @krupis first code and i'm having 2 problems:
> 
> 1. It's very slow, for downloading/saving a sketch of 340KB it takes over an hour.
> 2. It doesn't stop, it goes over 100%, i stopped it after 2 hours of downloading at about 175%.
> 
> I tried adding `if(readLength == contentLength) break;` in the downloading while loop, it took about 1:30 hours but it ended with an error.
> 
> Is it normal that it takes so much?
> 
> Did I missed something?
> 
> Thanks in advance.

That is quite strange. I have tested with simillar size of file and it took me no longer than 1 minute. Are you getting good signal power?

DP1000, 2022-03-02:
> That is quite strange. I have tested with simillar size of file and it took me no longer than 1 minute. Are you getting good signal power?

I'm printing `modem.getSignalQuality()` at every `printPercent()` and it's always at 31, that it should be the maximum signal strenght.

DP1000, 2022-03-02:
Ok, I've found out that something that i've added was interfering.
I'm still searching what I've done wrong, when i'll discover it i'll add to this comment, in case anyone will encounter the same problem.

Sorry for bothering.

ilker07, 2022-03-03:
> Ok, I've found out that something that i've added was interfering. I'm still searching what I've done wrong, when i'll discover it i'll add to this comment, in case anyone will encounter the same problem.
> 
> Edit: The problem was `#define TINY_GSM_RX_BUFFER 2048` instead of `#define TINY_GSM_RX_BUFFER 1030`, if someone knows why it's a problem let me know.
> 
> Sorry for bothering.

I never tried to increase rx buffer.

Sivapraveen-Entuple, 2022-03-24:
> > That is really weird. I have created an apache server using xampp and can do an OTA from there. What error are you getting? Are you using esp32 with spiffs fs partition? Do you know where your code gets stuck
> > […](#)
> 
> Now it worked! Thank you very much for making your code available and helping me. I changed the server and it worked. I was using 000webhost. And it was certainly my mistake, I was not sure how to correctly make the file available for download. Switching from server to another site / server with free hosting worked. Thank you very much for your attention in trying to help me. I wish success for your projects.

[eueduardo3](https://github.com/eueduardo3) I'm also trying to update firmware via OTA using SIM868. Could you please share you firmware which is working fine. it will be more helpful for me to further

thanks in advance

ilker07, 2022-03-28:
> [image]
>
> Should I change these settings?

How should these settings be? or What does it change?

Why  this is like this? Why 1030?  When will we change it?  #define TINY_GSM_RX_BUFFER 1030

krupis, 2022-03-28:
Default settings are good. You can change various things here but I suggest you keep them as they are untill you get more familiar with each one and then you know exactly what you are doing.

Its been a while but TINY_GSM_RX_BUFFER 1030 does not have to be 1030. You can play with it and will have an effect on how many bytes can modem receive at once. 

Actually, it should be set to 1024 because it would make more sense as this would mean you have 1kB chunks.

Either way, it should work :) are you having any problems?

ilker07, 2022-03-28:
> Default settings are good. You can change various things here but I suggest you keep them as they are untill you get more familiar with each one and then you know exactly what you are doing.
> 
> Its been a while but TINY_GSM_RX_BUFFER 1030 does not have to be 1030. You can play with it and will have an effect on how many bytes can modem receive at once.
> 
> Actually, it should be set to 1024 because it would make more sense as this would mean you have 1kB chunks.
> 
> You can try for yourself and set the TINY_GSM_RX_BUFFER to values such as ,128, 256, 512 ,1024, 2048 and etc and see what difference you see. Either way, it should work :) are you having any problems?

Actually right now ,i have no trouble at all .I just wondered why this numbers.I use buffer 1030.I just wanna know when we will change.Right now with this code it takes 10-15 minutes to update code.I never tried to change buffer value or partition scheme settings.Could it be related these 2 things?

krupis, 2022-03-28:
> > Default settings are good. You can change various things here but I suggest you keep them as they are untill you get more familiar with each one and then you know exactly what you are doing.
> > Its been a while but TINY_GSM_RX_BUFFER 1030 does not have to be 1030. You can play with it and will have an effect on how many bytes can modem receive at once.
> > Actually, it should be set to 1024 because it would make more sense as this would mean you have 1kB chunks.
> > You can try for yourself and set the TINY_GSM_RX_BUFFER to values such as ,128, 256, 512 ,1024, 2048 and etc and see what difference you see. Either way, it should work :) are you having any problems?
> 
> Actually right now ,i have no trouble at all .I just wondered why this numbers.I use buffer 1030.I just wanna know when we will change.Right now with my code it takes 10-15 minutes to update code.I never tried to change buffer value or partition scheme settings.Could it be related these 2 things that it takes 10-15 minutes?
> 
> For example my bin size 426 kb.how can we relate between this file size and buffer size?

10-15 minutes seems like quite a long time. The buffer size may have tiny affect on this but you definately have some other issues. When I did my testing, with good signal strength I managed to perform OTA in about 30 seconds. 

You should try different server to confirm if the issue is not from the server side and also ensure that you get good signal.

The buffer size simply describes what is the maximum receive buffer length. Since you now defined it to be 1030, that means the modem can receive 1030 bytes before it needs to split the chunk and receive another chunk of 1030 bytes.

ilker07, 2022-03-29:
> > > Default settings are good. You can change various things here but I suggest you keep them as they are untill you get more familiar with each one and then you know exactly what you are doing.
> > > Its been a while but TINY_GSM_RX_BUFFER 1030 does not have to be 1030. You can play with it and will have an effect on how many bytes can modem receive at once.
> > > Actually, it should be set to 1024 because it would make more sense as this would mean you have 1kB chunks.
> > > You can try for yourself and set the TINY_GSM_RX_BUFFER to values such as ,128, 256, 512 ,1024, 2048 and etc and see what difference you see. Either way, it should work :) are you having any problems?
> > 
> > 
> > Actually right now ,i have no trouble at all .I just wondered why this numbers.I use buffer 1030.I just wanna know when we will change.Right now with my code it takes 10-15 minutes to update code.I never tried to change buffer value or partition scheme settings.Could it be related these 2 things that it takes 10-15 minutes?
> > For example my bin size 426 kb.how can we relate between this file size and buffer size?
> 
> 10-15 minutes seems like quite a long time. The buffer size may have tiny affect on this but you definately have some other issues. When I did my testing, with good signal strength I managed to perform OTA in about 30 seconds.
> 
> You should try different server to confirm if the issue is not from the server side and also ensure that you get good signal.
> 
> The buffer size simply describes what is the maximum receive buffer length. Since you now defined it to be 1030, that means the modem can receive 1030 bytes before it needs to split the chunk and receive another chunk of 1030 bytes. In theory, if you had infinite amount of RAM, you could set the RX buffer to something ridiculous lets say 1MB. That would allow you the receive a whole firmware in a single data chunk thus saving some time in theory.. In practise things vary..

acttually you may be right .İt takes always 10-15 minutes whatever the size of the file.Could it be realated power issues or gprs?

krupis, 2022-03-29:
Yes. Power issues or the signal can be related to the length of the OTA.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ilker07, 2022-03-29:
> The buffer size simply describes what is the maximum receive buffer length. Since you now defined it to be 1030, that means the modem can receive 1030 bytes before it needs to split the chunk and [...]

> Yes. Power issues or the signal can be related to the length of the OTA. Ensure that the MODEM must have stable and powerful enough power supply. If you want to test that, You can provide an external power to the modem if you have a lab power supply unit

How do you supply the circuit?Can you show me?

krupis, 2022-03-29:
Could you show me what board are you using?

ilker07, 2022-03-29:
> Could you show me what board are you using and explain how are you currently powering it?

https://www.google.com/search?q=esp32+sim800l&sxsrf=APq-WBsg51fa-os3qjPGt3zWJpx_9w5bPg:1648534756303&source=lnms&tbm=isch&sa=X&ved=2ahUKEwjnk7LE1ur2AhVFSfEDHcPRA_MQ_AUoAnoECAEQBA&biw=1536&bih=760&dpr=1.25#imgrc=6YoIyMiuhT-t0M

ilker07, 2022-03-29:
> Could you show me what board are you using and explain how are you currently powering it?

3.7 v battery

krupis, 2022-03-29:
That is a first indicator that you have some power issues if it does not work when USB is disconnected. 
Can you measure the voltage across the battery terminals?
 Have you tried with 100% charged battery? 
Have you got a spare battery just to test it?

krupis, 2022-03-29:
Also, are you using battery connector that is located on the bottom of the board to connect your 3.7V battery or you are connecting it to 3.3V? 

The correct way is to connect it to battery connector

ilker07, 2022-03-29:
> That is a first indicator that you have some power issues if it does not work when USB is disconnected.
> Can you measure the voltage across the battery terminals?
> Have you tried with 100% charged battery?
> Have you got a spare battery just to test it?

Unfortunaletly i dont have spare battery.Right now i dont have the circuit but the voltage across the battery terminals around 4v as far as i remember.

ilker07, 2022-03-29:
> Also, are you using battery connector that is located on the bottom of the board to connect your 3.7V battery or you are connecting it to 3.3V?
> 
> The correct way is to connect it to battery connector

yeah,i use that way

krupis, 2022-03-29:
Seems to be a common power issue for these boards.. Have a read at:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/29

https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/27

If you have a lab power supply you can experiment with the power but if you dont know what you are doing you might destroy the device.

You could probably wire the lab power supply terminal wires to the usb +5V and GND pins. That way you would have a strong power supply ( much stronger than you PC USB ) . Keep in mind that the whole system probably needs up to 3A to be stable. So you need a power supply that can deliver 5V 3A.

ilker07, 2022-03-29:
> Seems to be a common power issue for these boards.. Have a read at: #29
> 
> [Xinyuan-LilyGO/TTGO-T-Display#27](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/27)
> 
> If you have a lab power supply you can experiment with the power but if you dont know what you are doing you might destroy the device.
> 
> You could probably wire the lab power supply terminal wires to the usb +5V and GND pins. You may want to disconnect the battery when you do this. That way you would have a strong power supply ( much stronger than you PC USB ) . Keep in mind that the whole system probably needs up to 3A to be stable. So you need a power supply that can deliver 5V 3A.

i have power supply 5v 3a .  i will try what you said.i will let you know ,Thanks.

[6 more replies]