Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #139: Version 1.4 with CH9102F doesn't work with Mac OS X

u-l-m-i, 2021-05-22:
Orderderd a replacement for my SIM800 1.4 with CP2104 and got a new one with CH9102F which fails to upload with the following error Message in PlatformIO on macOS 11.3.1 (OSX)

```
Serial port /dev/cu.usbmodem52780006671
Connecting....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:xx:xx:xx
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
*** [upload] Error 2_
```

The only difference I could spot between my old and the two new boards is the CH9102F Chip.

Any ideas, how to solve this issue?

azrin1972, 2021-05-23:
Hi, I'm having the same issue here

matymad, 2021-05-29:
Same issue here. With a 2015 iMac and 2017 Macbook Pro Touchbar both with MacOS Big Sur.

arjenhiemstra, 2021-06-01:
+1 for this issue, works with Windows without an issue. Same board is not working on MacOS (both Catalina and BigSur)

rynskyi, 2021-06-08:
+1 the same issue v1.4 with MacOS

SrRockabilly, 2021-06-13:
I got the same problem. While the new one with the cp120 is arriving I came to some questions. The device with the ch9102f came with a built in program that connected with the GPRS despite the fact that the program had not the apn. Im trying to connect with a sim800l modem and I cannot connect. I just could with my mobile phone and the ch9102f lily go. Weird isnt?
The prompt gave something like:
Found 37 networks
Begin SIM800 PASS
SIM800 Begin PASS
Test motor ...
Init success
AT+CHFA=1
AT+CRSL=100
AT+CLVL=100
AT+CLIP=1
etc

matymad, 2021-06-19:
I had to install Ubuntu 20.04 with VirtualBox on my iMac. The Ubuntu system has a little problem with the esptool.py v3.0-dev which is solved in this issue on github: https://github.com/espressif/esptool/issues/528

I exposed the Esp32 board to the virtual machine on the Devices -> USB menu an selected the device with name QuinHeng Electronics USB Single Serial. I ran Arduino IDE with sudo (in order to receive the serial monitor messages) and works fantastic.

There's images of the Arduino Terminal when compiling:
[image]

[image]

The board works fine WITHOUT having to jump the GPIO 0 and GND pins. 

NOTE: Works either with USB 2.0 and 3.0 controllers.

Hope it helps

matymad, 2021-06-26:
The driver for the CH9102F chip on MacOS is still under development, chinesse says:
http://www.wch.cn/bbs/thread-72035-1.html

genotix, 2021-08-23:
It's August in the mean time where they announced the driver to be there in May.
Also a MacOS user here; does anyone have a workaround for this one other than Parallels / Linux installation :)

genotix, 2021-08-24:
I just received this driver for the CH9102F from a Vendor.

Haven't tested it yet!!
[CH9102_Mac_Driver.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7037260/CH9102_Mac_Driver.zip)

mtgred, 2021-08-24:
@genotix The zip file contains CH34xVCPDriver_signed.pkg which seems to be the wrong driver (ie. CH34x instead of CH9102)

genotix, 2021-08-24:
> @genotix The zip file contains CH34xVCPDriver_signed.pkg which seems to be the wrong driver (ie. CH34x instead of CH9102)

Yes, I know; it is not. The CH34x driver and CH9xxx drivers have been sharing the same package for some time now.

mtgred, 2021-08-24:
It works on my Macbook Pro running on Big Sur 11.1

I initially tried on my iMac on Mojave 10.14.2 and that didn't work. It looks the drivers requires Big Sur.

Thanks a lot for the file @genotix !

marcelstoer, 2021-08-28:
Maybe the reason for specific driver for CH9102 to be missing is here: http://www.wch-ic.com/downloads/CH9102DS1_PDF.html (focus is mine)

> _**CH9102 supports built-in CDC driver in operating system**_ or multi-functional high-speed VCP manufacture driver

-> they don't "expect" a specific driver to be required and, hence, they don't provide one

jpduhen, 2021-09-03:
@genotix  Suggested driver does not work on my MacBook Pro (early 2015) - Big Sur 11.4.
Any help appreciated...

cgreening, 2021-09-03:
Doesn't work on Catalina either :(

genotix, 2021-09-03:
The only thing I can do is point you guys towards LilyGo shop on AliExpress.
It is where I bought mine and it seems to be the official LilyGo team behind it.

lilygo Official Store
https://a.aliexpress.com/_vCIogm

Do everyone a favour and should you receive any drivers from them, make them available to others.

marcelstoer, 2021-09-03:
> Suggested driver...

Which one do you refer to? The driver attached to https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139#issuecomment-904390716 does work for me. However, the driver linked to in [the later comment](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139#issuecomment-904982009) (the one from the vendor website) doesn't work for me either even though this one looks more official without the Chinese dialogs.

cgreening, 2021-09-03:
I seem to have got slightly further - now it's failing on

```
A fatal error occurred: Timed out waiting for packet header
```

marcelstoer, 2021-09-03:
The usual remedy for any sort of timing issues is reducing the upload speed. My driver/OS/device combination only supports max. 456300.

jdyoung76, 2021-09-24:
> I just received this MacOS Big Sur driver for the CH9102F chipset on the TTGO T-Call 1.4 from it's vendor.
> 
> I can hereby confirm this driver to be working! (Big Sur 11.5.2)
> 
> **Note 1:** You need to click through some Chinese screens
> **Note 2:** You need to select to correct device
> You will see:
> 
> 1. /dev/cu.usbmodemXXXXXXXX
> 2. /dev/cu.wchusbserialXXXXXXXX
> 
> Make sure to choose the second option!
> 
> [CH9102_Mac_Driver.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7037260/CH9102_Mac_Driver.zip)

This worked for me!  Big Sur 11.6 flashing TTGO LORA32 915MHz SX1276 OLED development boards!  I spent hours trying so many solutions!! This is the only thing that worked!!!!  I am so grateful I found this answer!!!!!!!!!!!!!!!!!!!

krazyeom, 2021-09-25:
> I just received this MacOS Big Sur driver for the CH9102F chipset on the TTGO T-Call 1.4 from it's vendor.
>
> I can hereby confirm this driver to be working! (Big Sur 11.5.2)
>
> **Note 1:** You need to [...]

This worked for me on Big Sur 11.6, too. thank you so much.

JayPalm, 2021-11-13:
This is what the chinese pop up says. Anyone have any thoughts about the safety of this, or how to find what was installed?

> Window
Launch the application and click Install driver, please complete the system driver authorization. To uninstall the driver, just transfer
Just move the application into the waste paper label. I feel that you are paying attention to Qinheng!
Install the driver

marcelstoer, 2021-11-13:
@JayPalm did you try whether the 1.5 version of their driver that has since been officially released at http://www.wch-ic.com/downloads/CH341SER_MAC_ZIP.html wouldn't solve the problem? I'd feel a little safer than installing some arbitrary file someone posted on GitHub.

cirobruno, 2021-11-19:
I've got the same problem.

Yesterday I've received a LilyGo TTGO T3_V1.6.1, that embeds this CH9102F.

Initially it used to show up as `/dev/cu.usbmodemFA1341`. It was able to output the demo code messages to the serial monitor. Nevertheless, unable to upload new code toward the microcontroller's flash memory. It used to report like this:

```
Serial port /dev/cu.usbmodemFA1341
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
*** [upload] Error 2
```

Then I’ve installed the driver found in [.](https://github.com/WCHSoftGroup/ch34xser_macos) It didn’t work out. Moreover, after installing it, the serial interface became invisible to VSCode’s serial monitor as well as for other serial monitors.

Then, I’ve tried to download [,](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7037260/CH9102_Mac_Driver.zip) but the provided file is called `CH34xVCPDriver_signed.pkg`, that doesn’t change anything!

I'm using Mac OS X 10.13.6 High Sierra.
Is there something else I should try?

Thank you.
Regards, Ciro

cirobruno, 2021-11-19:
Hello @Sladerix 

It looks like this link doesn't offer the same installer anymore. 

> > I just received this MacOS Big Sur driver for the CH9102F chipset on the TTGO T-Call 1.4 from it's vendor.
> > I can hereby confirm this driver to be working! (Big Sur 11.5.2)
> > **Note 1:** You need to click through some Chinese screens **Note 2:** You need to select to correct device You will see:
> > ```
> > 1. /dev/cu.usbmodemXXXXXXXX
> > 
> > 2. /dev/cu.wchusbserialXXXXXXXX
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Make sure to choose the second option!
> > [CH9102_Mac_Driver.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7037260/CH9102_Mac_Driver.zip)
> 
> Thanks! it solved the problem for LilyGo T-Call SIM800L AXP192 on Big Sur 11.6

I've just tried it and what I've got has been a file called `CH34xVCPDriver_signed.pkg`

Sladerix, 2021-11-20:
> Hello @Sladerix
>
> It looks like this link doesn't offer the same installer anymore.
>
>
> I've just tried it and what I've got has been a file called `CH34xVCPDriver_signed.pkg`

Hi @cirobruno actually i went in my downloads and i found what i downloaded that day:
[CH9102_Mac_Driver.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7574304/CH9102_Mac_Driver.zip)

But if you unzip it you will find the same file you mentioned..

philwinder, 2022-01-06:
For the benefit of future googlers, below is a link to download the correct driver for the `CH9102F` chip for OSX direct from the manufacturer:

http://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html

Once installed, the device will be listed under `tty.wchusb...` or `cu.wchusb...`

Please note that the link posted in https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139#issuecomment-967794913 is not correct. This driver is for the older CH341 chip, not the CH9102F.

Other platforms are available from here: http://www.wch.cn/search?q=CH9102&t=downloads

Thanks, Phil

dennisdebel, 2022-01-06:
Dear Phil, 
Thank you for the correct link, I am on Catalina and the driver you posted seemed to work for CH9102X. However...only one time, after flashing firmware to the board, the port disappears. No clue how to get it back besides uninstalling, reinstalling and rebooting.

oojBuffalo, 2022-01-17:
The latest driver (updated 2021-11-11) from the official website works with macOS 12.1 Monterey as well.

flidsky, 2022-01-17:
Guys I solved the problem on Monterey this way:
1) Install ch34x driver
2) Press mysterious chinese button
3) GO TO Security on Settings and allow chinese stuff to open
usbmodem changes to wchusbmodem

SchneHa, 2022-01-22:
I found a solution for my Catalina based Mac. It's now working with CH9102 devices and V1.7 of WCH driver. I did the following steps (look [here](https://forum.m5stack.com/topic/3760/cant-flash-latest-m5-stamp-c3/21)

zeropool, 2022-02-18:
> This worked for me! Big Sur 11.6 flashing TTGO LORA32 915MHz SX1276 OLED development boards! I spent hours trying so many solutions!! This is the only thing that worked!!!! I am so grateful I found [...]

pepotx, 2022-03-24:
> For the benefit of future googlers, below is a link to download the correct driver for the `CH9102F` chip for OSX direct from the manufacturer:
> 
> http://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html
> 
> Once installed, the device will be listed under `tty.wchusb...` or `cu.wchusb...`
> 
> Please note that the link posted in [#139 (comment)](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139#issuecomment-967794913) is incorrect. This driver is for the older CH341 chip, not the CH9102F.
> 
> Other platforms are available from here: http://www.wch.cn/search?q=CH9102&t=downloads
> 
> Thanks, Phil

Worked for me in Yosemite.
Thank you very much!

shihchuansun, 2022-04-15:
I tried all the recommendations, but I can't seem to make it work on Monterey 12.1 on M1 Pro

cgreening, 2022-04-15:
I found that I had some old silicon lab drivers that were interfering. Unfortunately I can't remember the exact steps I used to clean it all up as I did so much messing around.

RafBois, 2022-07-15:
I tried to contact tech@wch.cn them but the email gave me an error in return. 

The ARM technologies is the future is a lack of knowledge or a simple problem of profitability or the 2 maybe?

KenjutsuGH, 2022-07-17:
I have been successful in using the TTGO T5 e-Paper modules (with CH9102F) by connecting them to a **_USB 2.0_** hub and then to the USB 3 hub to the M1 Air. I do not have a USB 2.0 hub with a USB C cable to directly connect to the M1. So far I have had a 100% success rate in connecting the T5 displays via a USB 2 hub

SebastianKumor, 2022-08-10:
> For the benefit of future googlers, below is a link to download the correct driver for the `CH9102F` chip for OSX direct from the manufacturer:
>
> http://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html [...]

This worked for me also on monterey 12.5

ErickGiffoni, 2022-09-07:
Hi,

I am having a similar problem here with macOS **Big Sur 11.5.2**

It's similar because I'm not using a CH9102F, but a **CH9102X**.

I'm using the PlatformIO vsCode extension to upload code to my **ESP32**, so I get the error:
```
Auto-detected: /dev/cu.usbmodem539E0028411
Uploading .pio/build/esp32doit-devkit-v1/firmware.bin
esptool.py v3.3
Serial port /dev/cu.usbmodem539E0028411
Connecting....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:dc:7e:62:df:d4
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
*** [upload] Error 2
```

Any clues ?
Tks

moyun-yinyun, 2022-09-08:
> Hi,
>
> I am having a similar problem here with macOS **Big Sur 11.5.2**
>
> It's similar because I'm not using a CH9102F, but a **CH9102X**.
>
> I'm using the PlatformIO vsCode extension to upload code to [...]

1. If you are currently using the kext driver, please delete the driver and reinstall the new driver.
cd /Library/Extensions 
sudo rm -rf CH34xVCPDriver.kext
sudo rm -rf CH34xVCPControlDriver.kext

2. The CDC driver that comes with macOS is prone to errors during data communication due to the incomplete protocol. It is not recommended to use the macOS CDC driver.

3. For macOS 11.0, the VCP driver of WCH needs to be installed to make the device work. Please refer to the download link for installation.

https://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html

Best regards

maxakash, 2022-09-18:
> I just received this MacOS Big Sur driver for the CH9102F chipset on the TTGO T-Call 1.4 from it's vendor.
>
> I can hereby confirm this driver to be working! (Big Sur 11.5.2)
>
> **Note 1:** You need to [...]

Worked for me on M1 Max MacOs Monterey. Thanks

icenpotatos, 2024-04-21:
> I just received this MacOS Big Sur driver for the CH9102F chipset on the TTGO T-Call 1.4 from it's vendor.
>
> I can hereby confirm this driver to be working! (Big Sur 11.5.2)
>
> **Note 1:** You need to [...]

Might've been an issue on my side, but this didn't work for me on Ventura 13.6.3. According to the popup, the driver needs to be updated.

Instead I found one that did work for me:
https://www.wch-ic.com/downloads/CH34XSER_MAC_ZIP.html 

and I followed the tutorial on this page (the zip file in this website did not work for me, but the WHC link did):
https://learn.adafruit.com/how-to-install-drivers-for-wch-usb-to-serial-chips-ch9102f-ch9102/mac-driver-installation

rogeriocassares, 2024-06-16:
THAAAAAAAANKS! ONE DAY ISSUE

tb205gti, 2024-07-07:
> 1. If you are currently using the kext driver, please delete the driver and reinstall the new driver.
>    cd /Library/Extensions
>    sudo rm -rf CH34xVCPDriver.kext
>    sudo rm -rf [...]

THIS worked!

Thank you.