Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #145: I cannot get LILYGO® TTGO T-Call  to work I lost LED flashing an board is stuck in a loop

chiggs002, 2021-06-27:
**Sketch. uploaded : I tried to run Tiny GSM SKETCH HttpClient uploaded.**

**Orange debugging Arduino console:**

Sketch uses 223482 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 14320 bytes (4%) of dynamic memory, leaving 313360 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port /dev/cu.usbserial-01E07DF1
Connecting........_
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c4:4f:33:54:36:1d
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5808.3 kbit/s)...
Hash of data verified.
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 0.2 seconds (effective 884.5 kbit/s)...
Hash of data verified.
Compressed 223600 bytes to 116553...
Writing at 0x00010000... (12 %)
Writing at 0x00014000... (25 %)
Writing at 0x00018000... (37 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (62 %)
Writing at 0x00024000... (75 %)
Writing at 0x00028000... (87 %)
Writing at 0x0002c000... (100 %)
Wrote 223600 bytes (116553 compressed) at 0x00010000 in 1.9 seconds (effective 945.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 2607.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

serial monitor loop stuck:** **STUCK IN A REBOOT [LOOP:]([url](url))

Wait...
[34] Trying baud rate 115200 ...
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d1bdc: 01ca6520 4e6e666c 4444f45d
Core 1 register dump:
PC      : 0x400d1be2  PS      : 0x00060730  A0      : 0x800d2290  A1      : 0x3ffb1f10  
A2      : 0x3ffc01f0  A3      : 0x3ffc020c  A4      : 0x0001c200  A5      : 0x00000000  
A6      : 0xffffffe6  A7      : 0x0000000a  A8      : 0x800d1be0  A9      : 0x3ffb1ef0  
A10     : 0x00000105  A11     : 0x3f4002fe  A12     : 0x0800001c  A13     : 0x00000009  
A14     : 0x00000009  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffbe  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d1be2:0x3ffb1f10 0x400d228d:0x3ffb1f70 0x400d44fa:0x3ffb1fb0 0x400860ed:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Wait...
[34] Trying baud rate 115200 ...
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d1bdc: 01ca6520 4e6e666c 4444f45d
Core 1 register dump:
PC      : 0x400d1be2  PS      : 0x00060730  A0      : 0x800d2290  A1      : 0x3ffb1f10  
A2      : 0x3ffc01f0  A3      : 0x3ffc020c  A4      : 0x0001c200  A5      : 0x00000000  
A6      : 0xffffffe6  A7      : 0x0000000a  A8      : 0x800d1be0  A9      : 0x3ffb1ef0  
A10     : 0x00000105  A11     : 0x3f4002fe  A12     : 0x0800001c  A13     : 0x00000009  
A14     : 0x00000009  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffbe  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d1be2:0x3ffb1f10 0x400d228d:0x3ffb1f70 0x400d44fa:0x3ffb1fb0 0x400860ed:0x3ffb1fd0

**See attachments of photos of device, errors.in files
• MY_PHOTOS_LilyGo-T-Call-SIM800
• deugging_info.RTF
•serial_monitor error.RTF**
[]([](https://drive.google.com/drive/u/1/folders/1LZJ6GYOwREr-m1xoGWPurZkp_vR38zeg))

https://drive.google.com/drive/u/1/folders/1LZJ6GYOwREr-m1xoGWPurZkp_vR38zeg

**Why has the SIM800L device stopped SIGNALLING ME WITH THE LED? LED STOPPED FLASHING? ( EVEN IF I LOAD AN EMPTY SKETECH) WHERE IS THE DEFAULT SKETCH THAT WAS PROGRAMMED INTO IT ?**

mateusfrizon, 2021-08-31:
i have the same problem with more than one board. Can someone helppp

Haasrobertgmxnet, 2021-09-14:
I have also the same problem with two of these boards.

mateusfrizon, 2021-10-19:
> Hello, does this apply to uploading other examples?

i just solved the problem, i guess that it was one of the sim800 pins that
we put on the prototip board, but wasn't used, is possible that some pin
can get interfered just for putting on a connector ? because the problem
was solved after removing from the conector the unused pins.