Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #146: Power supply

Michel-Mayoud, 2021-06-28:
I would like to power the SIM800C AXP192 through the back board connector. Unfortunatly, it doesn't work : the baord starts and turn off after a few seconds. I tried to press the reset button without any beter result. I use a Li-ion batterie 3.7V. Is it a normal running ?
Thanks a lot for your return

komkritc, 2021-06-28:
I got similar issue here. turn on with battery then green led (io13) on and then off. 3V3 pin voltage is only 0.9V
No response from any button

Michel-Mayoud, 2021-06-30:
Hi,
I've done a similar observation (battery power on back connector) :
1 Battery 3.3V on
2 Green led on
3 Program start (I can see that because of a witness led blink IO02)
4 Green led off
5 The 3.3V output shut down to 0V

Did you find an idea to solve the problem ?

Best regards
Michel MAYOUD
06 37 38 82 79

komkritc, 2021-06-30:
I got exactly problem on model sim800c that I think came from defected board or poor soldering connection on axp192. I reheat with hot air gun. Now I use only model sim800L instead. My current problem is sometime the board cannot wakeup from deep sleep due to no 3.3v after wakeup.

komkritc, 2021-06-30:
to temporary fix my problem I use **light sleep** and this just working for now.

Michel-Mayoud, 2021-07-01:
Hi,
Thank you for your help.
I could find a temporary solution based on a 5V power supply applied on the
5VIN. I tried first with a Li battery and got the same issue. I tried
secondly adding a 4700µF capacitor (on the battery) and could execute the
whole program.

My problem with the SIM800L is that it doesn't work when powered through
the back connector (3.7 Li battery). I think I should test a Li battery
supported by a capacitor ?

About the sleep mode I did not observe a waking up issue, I use timed sleep
with functions : esp_deep_sleep_start and esp_sleep_enable_timer_wakeup,

Best regards
Michel MAYOUD
06 37 38 82 79

ignitionelectronics, 2021-11-07:
Hii, You Just need to send command to IP5306 for stay On for that you can check Example code AT debug Test

Michel-Mayoud, 2021-12-08:
> Hello, have you solved your problem?I'm powered by a lithium battery,
> which works fine.Is it more about the quality of lithium batteries?

Hello,
Thank you very much for your message. I could find a solution for my power
supply issue (simply use a lithium battery connected on the back board
connector).
I built a prototype which had been placed outdoors for a test periode
(winter conditions : wet and temperature around 2°C). Unfortunately, after
a couple of weeks it stopped working.No apparent reason, no damage, the
board doesn't work anymore. On power setting up, the power led is on for a
few seconds then turns off, the result is the same with USB power supply.
It is also impossible to upload a program, the USB port doesn't work.
Could you help me ?
I' working on a weight scale device used for beekeeping. This unsuccessful
test could encourage me to make the decision of using another board, even
if that one seems to be appropriate.
Please let me know your thoughts.

Best regards
Michel MAYOUD
+33 6 37 38 82 79

LilyGO, 2021-12-10:
1. Are there any ports now?
2. If there is a port. What does the serial port print?
3. What errors are reported when firmware is uploaded?
4. Can you make a video to show me the phenomenon?

ignitionelectronics, 2021-12-10:
Hi,

Sorry for the late reply

First of all connect module with your PC and change working or not if not
then may be your module is faulty.
if you module stop working after some time then give power from external 5v
and at 5V and gnd pin after that upload code.

Michel-Mayoud, 2021-12-10:
Hello,
I've just seen your last message.
I prepared some videos that I will send to you.
I will test the 5V power supply as soon as possible.

Best regards
Michel MAYOUD
06 37 38 82 79

Michel-Mayoud, 2021-12-10:
Hello,
Thank you for your return,

   - I do not use any other port than the USB portI can't upload a
   firmware, the ...
   - Here is the error on upload attempt : ...
   - You will find two video
   - The first with battery power supply
   - The second with USB power supply

The board ran correctly for two weeks, the battery was charged by the means
of a solar charger. You will find in three mails :
- TTGO BACK BOARD POWER SUPPLY.mp4 : The power led turns off after a few
seconds
- TTGO USB power supply.mp4 : same problem (second mail)
- TTGO after Power led off.mp4 : USB power supply, power led off the SIM
module turns on but the network connection fails (third mail)
Michel MAYOUD
06 37 38 82 79

Michel-Mayoud, 2021-12-10:
Second mail,

Michel MAYOUD
06 37 38 82 79

Michel-Mayoud, 2021-12-10:
third mail,

Michel MAYOUD
06 37 38 82 79

ignitionelectronics, 2021-12-10:
Where is the video i think you forgot to attach the video

Michel-Mayoud, 2021-12-10:
I'm surprised, checking my sent mail, the file seems to be in it.
I send it to you again.

Michel MAYOUD
06 37 38 82 79

ignitionelectronics, 2021-12-10:
Please send me mail at ***@***.***

Michel-Mayoud, 2021-12-10:
Sorry but I don't understand, mail seems to have been replaced by
"***@***.***"

Michel MAYOUD
06 37 38 82 79

ibrcic, 2021-12-10:
> Hello, have you solved your problem?I'm powered by a lithium battery, which works fine.Is it more about the quality of lithium batteries?

I have exactly the same problem. The SIM800C cannot connect to the network when board is powered through back connector with battery. The status LED of modem lights up, it tries to establish the connection but then it just resets the board. This is clearly some form of brownout issue but I have high quality, brand new Samsung 18650 battery that should be able to provide up to 10A of current.

Additionally I tried to solder thicker wires to JST connector and to battery directly but I still have the same problem. I simply can't get SIM800C to power up and fully connect over battery. When connected through PC USB everything works as expected, no power issues at all. Can you recommend some troubleshooting steps because I am simply out of ideas...

ignitionelectronics, 2021-12-10:
Okay try to to change JST cables may be issue with cable.

ibrcic, 2021-12-10:
Yes, I already tried that. I soldered new, thicker wires directly to positive and negative terminal of JST connector and connected them directly to battery, but it didn't help. Here's the photo:

[image]

ignitionelectronics, 2021-12-10:
Where is the antenna ??

ibrcic, 2021-12-10:
You can see it on the image:

[image]

ignitionelectronics, 2021-12-10:
With usb working fine with this antenna right ?

ibrcic, 2021-12-10:
Yes, when I connect USB everything works as expected.

ignitionelectronics, 2021-12-10:
I think your boost converter shutdown after some time so just goto Google
and search for ttgo esp32 sim800l and download example from that copy code
for turning on boost converter.

ibrcic, 2021-12-10:
Is that something that I need to do even if I have never revision with AXP192 on board? I've seen boost converter mentioned in context of older revisions that have IP5306 on it.

ignitionelectronics, 2021-12-11:
Yeah.
please check

proximus323, 2021-12-12:
Did some one figure out issue with battery on SIm800L ? 
I have 3 boards and all 3 not working on battery only USB, 
Is thera a option to connect 5V to USB ?

Michel-Mayoud, 2021-12-12:
Hello,
From my point of view, you should use an lithium battery with an output current hight enough. My experiment : I used a battery with a current limitation inegrated system (9900 mA, not stated on amazon description) and as the SIM800 tried to make a connection the voltage droped to a insuffisant level creating disfunctions. I found another battery making sure that the output current was high enough and the board works using the back board connector.
One more detail, I could observe that using a bread board and the wires adapted for the voltage droped, so make sure your connection battery - board is as short as possible.

ibrcic, 2021-12-12:
I was able to make it work when I fully charged my battery to 4.2V, then I tried testing to see when will SIM8000C stop working and it happened when battery discharged to about 3.9V. Because of this I can only get about 50-60% of total usable battery charge.
I have no idea why this is happening. @Michel-Mayoud I see that you mentioned that it was a problem because battery was limited to 9900mA. My battery is limited to 10A also but that should be plenty enough for peek currents that are up to 2A for SIM800C if I remember correctly.

proximus323, 2021-12-12:
This board sucks, I put 4.2 V from the power supply, it start working but for about 30 second voltage for 5V PIN got to 2.25 V and 3.3V PIN is now 0V, ESP32 not wake up from deep sleep, ith only work on USB port.
This can NOT be used in any projects.

ibrcic, 2021-12-12:
What board do you have? AXP192 or IP5306?

proximus323, 2021-12-12:
IP5306

ibrcic, 2021-12-12:
Did you turn on power boost? This is something you have to do and is already done in examples in utilities.h. If you for some reason don't have utilities.h in your project, this is how you turn on boost.

```
bool setupPMU()
{
  bool en = true;
  Wire.begin(I2C_SDA, I2C_SCL);
  Wire.beginTransmission(IP5306_ADDR);
  Wire.write(IP5306_REG_SYS_CTL0);
  if (en)
  {
    Wire.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  }
  else
  {
    Wire.write(0x35); // 0x37 is default reg value
  }
  return Wire.endTransmission() == 0;
}
```

proximus323, 2021-12-12:
I use esp-idf, I will check this, thanks.

proximus323, 2021-12-13:
Ok, a spend cople of hours to set this value, in for now is working fine on 18650 battery.

ibrcic, 2021-12-13:
That's great! Unfortunately that does not work on newer revision with AXP192 and I still can't get it to work when battery voltage drops below about 3.9V. Does anyone have a solution for that on AXP192 board?

Michel-Mayoud, 2021-12-13:
Hi,
From my experiment, the board stops working when battery voltage drops to about 3.5V. Perhaps should you try more powerfull battery  ?

ibrcic, 2021-12-13:
I have Samsung 18650 22P which, according to the data sheet, should be able to allow up to 10A discharge. That should be more than enough.

Here is the datasheet. https://www.tme.eu/Document/cf2b91a94627173083963b831db6ea62/ACCU-18650-2.2P-HV.pdf

When I added 200uF capacitor I was able to get to 3.8V, I guess I could try adding even larger capacitor, maybe 1000uF? But still, strange that the board doesn't work properly with the battery out of the box

uvce12, 2022-02-03:
I had same problem for a while.

I found a really old 3.7V LI-PO battery lying in the workshop.

It works perfectly fine now. I exactly don't know why. But LI-Po looks like is the only fix for this problem as of now.

realbeeit, 2022-02-13:
Hello, did anyone solved this yet? I also have problem with SIM800C_AXP192_VERSION_20200609. On USB is running fine, on battery it crash on modem start up. Battery is fully charged. @ibrcic you are from Croatia, so am I, please contact me directly on my info at beeit . hr mail if you want.

Michel-Mayoud, 2022-02-14:
Hello,
Yes, the TTGO SIMC APX wpks fine by using a battery able to deliver more then 2 Amps peak current. In my first tests, the batterie I used (3000mA) integrated an internal current limitation, so take care about the battery you choose. Anyway, the power supply is still the main problem of this board. 
Best regards

realbeeit, 2022-02-14:
I am using 10 000mAh lithium ion battery. I managed it to turn on ok if I connect it directly, but crashing if connecting through solar charger. I do not know why...I measured the current using scope and max current was about 250mA.

Michel-Mayoud, 2022-02-14:
You have to make shure that your battery accept hight peak current. I analysed the current used by the TTGO board and I clearly identified the peak current about 2 Amps. The battery I used first had an internal limitation, make sure you are not in this case and keepp in mind the battery peak current (mA) is different of the capacity (mAh).

realbeeit, 2022-02-14:
Yes I understand difference between capacity and peak current. I think that something is blocking in solar charger output to the board. I won't be using 20200609 boards anymore, as I said, 20200327 is working perfectly and the difference between them is SIM800C and SIM800L.

ibrcic, 2022-02-14:
Make sure that when you connect the solar board to TTGO you do so through 5VIN PIN and not the Battery connector on the back of the board, as that wouldn't accept larger V of solar charger. Anyway, I am not even using solar charge controller, I just connected photovoltaic cell directly to the 5VIN and let the board IC control the battery charge.

But, latest revision of the board seems to have some issues when the battery V drops bellow about 3.9V. This is not the problem of peak current as I use high quality Samsung and Panasonic batteries capable of providing up to 10A and other users even tried powering the board with the PSU. It now seems that this is a hardware design problem as the maker of the board recently commented on another issue that they will be changing the design of power circuit. Here's the issue: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/189

To remedy this problem it is possible to add capacitor (I added 470uF), and that will somewhat help, but will not completely resolve the problem. I also read that some users used voltage boosters and then connected the output to 5VIN as opposed to battery connector. That should also work but then you need separate circuit to prevent over discharge (and overcharge if you are using solar) of the battery, and that defeats the purpose of the board's all in one AXP192 that should take care of that.

@realbeeit you can contact me to my mail ibrcic [at] gmail.com.

realbeeit, 2022-02-14:
I will try that today. Only I will connect battery directly to battery connector, and solar panel to 5VIN because as I read in datasheet that is connected to 5V USB bus. My board is working even if its battery is charged to 3.6v via battery connector. So If I can connect solar panel directly to 5VIN and board charge battery then my problem is solved.

ceren16, 2023-08-09:
> ith only work on USB port

did you find solution ? does it only work on USB port?

Atomicix26, 2024-05-02:
สวัสดีครับพี่เคน พวกผมรับผิดชอบ Task1 และ 9 ก่อนอื่นขอรายงาน Task1 ครับ ว่าเป็นยังไงบ้าง คืเราได้เริ่มทดสอบอุปกรณ์ที่เราต้องการก่อนใช้งานแล้ว ซึ่งสามารถใช้งานได้ปกติ และ พวกเราได้เรี่มวางแผนการต่ออุปกรณ์,ส่วนอุปกรณ์ที่เราต้องการเพี่มนั้นน่าจะไม่มีแล้ว ที่เหลือคือการเชื่อม และ การต่ออุปกรณ์ตัวจริง 
[image]