Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #156: Store USART data in a buffer (SMS)

krupis, 2021-07-23:
Hello. In my program, I require to be able to change various program variables through the SMS. For example temperature, delay times and etc.

I am thinking of storing an SMS message to the buffer and then comparing to some specific string in my code. If it matches, then I change the setting, if it does not match, do nothing. For example, I send an SMS: "CTH:25" which would correspond to Change temperature high to 25 degrees. Then in my program I would change the variable to 25.

However, I am having a lot of issues saving the usart data to the buffer. 
I have configured the AT+CNMI = 1,2,0,0,0 so the message is routed directly to uart.

If I call  the following code in void loop, I can see the messages nicely
```
 if (SerialAT.available()) {
      Serial.write(SerialAT.read());
    }
    if (Serial.available()) {
      SerialAT.write(Serial.read());
    }
```

```
07:43:43.702 -> waiting for response complete
07:43:44.654 ->
07:43:44.654 -> Call Ready
07:43:45.401 ->
07:43:45.401 -> SMS Ready
07:43:50.677 ->
07:43:50.677 -> *PSUTTZ: 2021,7,23,4,43,49,"+12",1
07:43:50.677 ->
07:43:50.677 -> DST: 1
07:43:52.784 ->
07:43:52.784 -> *PSUTTZ: 2021,7,23,4,43,52,"+12",253
07:43:52.784 ->
07:43:52.784 -> DST: 253
07:43:52.784 ->
07:43:52.784 -> +CIEV: 10,"24601","Telia LT","Telia LT", 0, 0
```

However, I am now trying to do this in a different way:

```
char full_message[50];
byte index5;

void read_SMS_data(){
    if(SerialAT.available() > 0){
        index5 = 0;
        while(SerialAT.available() > 0)
        {
            char inChar = SerialAT.read();
            full_message[index5] = inChar;
            index5++;

        }
        Serial.print("full message = ");
        Serial.println(full_message);
    }
}
```

When I check the data in the serial monitor, it looks very strange:

```
07:50:28.142 -> waiting for response complete
07:50:29.121 -> full message =

07:50:29.121 -> full message =
07:50:29.121 ->
07:50:29.121 -> full message = C
07:50:29.121 -> full message = a
07:50:29.121 -> full message = l
07:50:29.121 -> full message = l
07:50:29.121 -> full message = 
07:50:29.121 -> full message = R
07:50:29.121 -> full message = eady
07:50:29.121 ->
07:50:29.872 -> full message =
ady
07:50:29.872 ->
07:50:29.872 -> full message =
07:50:29.872 -> ady
07:50:29.872 ->
07:50:29.872 -> full message = Sady
07:50:29.872 ->
07:50:29.872 -> full message = Mady
07:50:29.872 ->
07:50:29.872 -> full message = Sady
07:50:29.872 ->
07:50:29.872 -> full message =  ady
07:50:29.872 ->
07:50:29.872 -> full message = Ready
07:50:29.872 ->
07:50:29.872 -> full message =
07:50:29.872 -> ady
07:50:29.872 ->
07:50:33.606 -> full message =
07:50:33.606 -> ady
07:50:33.606 ->
07:50:33.606 -> full message =
07:50:33.606 ->
07:50:33.606 -> ady
07:50:33.606 ->
07:50:33.606 -> full message = *
07:50:33.606 -> ady
07:50:33.606 ->
07:50:33.606 -> full message = P
07:50:33.606 -> ady
07:50:33.606 ->
07:50:33.606 -> full message = S
07:50:33.606 -> ady
07:50:33.606 ->
07:50:33.606 -> full message = U
07:50:33.606 -> ady
07:50:33.606 ->
07:50:33.606 -> full message = TTZ:
07:50:33.606 ->
07:50:33.606 -> full message = 2021,7,23,4,50,32,"+12",1
07:50:33.606 -> full message =
07:50:33.606 ->
07:50:33.606 -> DST: 1
07:50:33.606 -> 50,32,"+12",1
07:50:35.566 -> full message =
07:50:35.566 ->
07:50:35.566 -> DST: 1
07:50:35.566 -> 50,32,"+12",1
07:50:35.566 -> full message =
07:50:35.566 ->
07:50:35.566 ->
07:50:35.566 -> DST: 1
07:50:35.566 -> 50,32,"+12",1
07:50:35.566 -> full message = *
07:50:35.566 ->
07:50:35.566 -> DST: 1
07:50:35.566 -> 50,32,"+12",1
07:50:35.566 -> full message = P
07:50:35.566 ->
07:50:35.566 -> DST: 1
07:50:35.566 -> 50,32,"+12",1
07:50:35.566 -> full message = SUTTZ: 2021,7,23,4,50,35,"+12",253
07:50:35.566 ->
07:50:35.566 -> DS
07:50:35.566 -> full message = T: 253
07:50:35.566 ->
07:50:35.566 -> +CIEV: 10,"24601","Telia LT","Telia LT", 0, 0
07:50:35.566 ->
```

It sometimes thinks that there is a null termination or end line character so it does not manage to append a full string. That is an issue for me. Could someone suggest a way to solve this problem?

krupis, 2021-07-23:
**UPDATE:**

I have managed to find a solution, I dont know if that is correct, but I was experimenting and notice that once I added some delay, it started to work better. I have added a 1ms delay in between each character read:

```
void read_SMS_data(){
    if(SerialAT.available() > 0){
        index5 = 0;
        while(SerialAT.available() > 0)
        {
            char inChar = SerialAT.read();
            delay(1);
            full_message[index5] = inChar;
            index5++;
        }

        Serial.print("full message = ");
        Serial.println(full_message);
    }
}
```

Now it seems that it is able to read a full message without any issues:
```
08:11:43.811 -> waiting for response complete
08:11:44.790 -> full message = 
08:11:44.790 -> Call Ready
08:11:44.790 -> 
08:11:45.536 -> full message = 
08:11:45.536 -> SMS Ready
08:11:45.536 -> 
08:11:45.536 -> 
08:11:50.435 -> full message = 
08:11:50.435 -> *PSUTTZ: 2021,7,23,5,11,49,"+12",1
08:11:50.435 -> 
08:11:50.435 -> DST: 1
08:11:50.435 -> 
08:11:52.536 -> full message = 
08:11:52.536 -> *PSUTTZ: 2021,7,23,5,11,52,"+12",253
08:11:52.536 -> 
08:11:52.536 -> DST: 253
08:11:52.536 -> 
08:11:52.536 -> +CIEV: 10,"24601","Telia LT","Telia LT", 0, 0

```

Can anyone explain why this happens ?

krupis, 2021-07-23:
I have been experimenting it further. Even adding 1ms delay does not seem to guarantee the correct message received:

I have tried to send 3 messages from my phone to the GSM and see how the serial monitor responds:

```
08:35:42.131 -> full message = 
08:35:42.131 -> +CMT: "+my_number","","21/07/23,08:35:38+12"
08:35:42.131 -> Sup
08:35:42.131 -> CIEV: 10,"24601","Telia LT","Telia LT", 0, 0
08:35:42.131 -> 

08:36:01.928 -> full message = 
08:36:01.928 -> +CMT: "+my_number","","21/07/23,08:35:58+12"
08:36:01.928 -> Hello
08:36:01.928 -> EV: 10,"24601","Telia LT","Telia LT", 0, 0
08:36:01.928 -> 

08:36:26.874 -> full message = 
08:36:26.874 -> +CMT: "+my_number","","21/07/23,08:36:22+12"
08:36:26.874 -> Hello this is message from my phone
08:36:26.874 -> ia LT", 0, 0
08:36:26.874 -> 

```
As you can see, first 2 messages seem okay, but the third message is missing some data on the last line:

isntead of EV: 10,"24601","Telia LT","Telia LT", 0, 0 I only received ia LT", 0, 0

LilyGO, 2021-10-19:
If no delay is added, the processing speed is instantaneous, which may cause data loss and assignment failure