Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #157: Can IP5306 be used for HTTPS request

ibrcic, 2021-07-31:
I have been trying on and off for months to make a HTTPS request with this model without a success. Upon searching the internet I am finding conflicting information on can this modem actually be used to make HTTPS (as opposed to HTTP) requests or not.

Has anyone been able to do it? If yes, how? I think I tried every possible solution suggested on the web, going as far as trying to do it directly with AT commands (without much luck). I am able to make http requests without a problem, but not https.

Are newer models improved in this regard?

I would appreciate any help with this.

LilyGO, 2021-09-13:
> I have been trying on and off for months to make a HTTPS request with this model without a success. Upon searching the internet I am finding conflicting information on can this modem actually be [...]

[Hello, this is a new example, you can refer to it.](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_HttpsClient)

ibrcic, 2021-09-14:
Hello, thank you for providing new example. Unfortunately I am still unable to connect with the example. Here are AT Commands, I am not sure what is wrong.

```
OK
Network connected
Connecting to internet.ht.hrAT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","internet.ht.hr"

OK
AT+CGDCONT=1,"IP","internet.ht.hr"

OK
AT+CGACT=1,1

+CMTI: "SM",9

+CMTI: "SM",10

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.116.118.28"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="internet.ht.hr","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

10.116.118.28

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 success
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

10.116.118.28

OK
GPRS connected
Performing HTTPS GET request... AT+HTTPTERM

+CME ERROR: operation not allowed
[60680] +HTTPTERM
AT+SAPBR=0,1

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.116.118.28"

OK
AT+HTTPINIT

OK
AT+HTTPPARA="CID",1

OK
AT+HTTPPARA="URL","www.baidu.com"

OK
AT+HTTPPARA="REDIR",1

OK
AT+HTTPSSL=1

OK
AT+HTTPACTION=0

OK
AT+HTTPREAD

OK
AT+HTTPTERM

+CME ERROR: operation not allowed
[73742] +HTTPTERM
AT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

+HTTPACTION: 0,601,0

OK
GPRS disconnected
```

LilyGO, 2021-10-15:
[Reference SIM800 SSL application Manual](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_HttpsClient/SIM800%2BSeries_SSL_Application%2BNote_V1.00.pdf)

LilyGO, 2021-10-23:
[Reference](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_HttpsClient)

ibrcic, 2021-11-25:
Linked reference does not mention error 601 anywhere as far as I can see. I bought another T-Call, and this time the latest one with SIM800C, and it still doesn't work. I get the same error 601 with the code you provieded.
I don't know why I can't seem to do HTTPS request...

LilyGO, 2021-11-26:
The default URL is www.baidu.com. Try using a different URL?

ibrcic, 2021-11-30:
Hi, yes, changing to www.google.com seems to be working, thank you! Now just to realize why doesn't ArduinoHttpClient library work, but I guess I have to check that with authors of the library.

Thank you again.

ibrcic, 2021-11-30:
Just one more question, why do you have  this `delay(10000)` in the GET example?

```
    modem.sendAT(GF("+HTTPACTION=0"));//Get
    if (modem.waitResponse(60000L) != 1) {
       DBG(GF("+HTTPACTION=0"));
        return false;
    }
    delay(10000);
```

I understand that without it, it doesn't work, but I am not sure why?

RevoltaAmadeus, 2023-06-06:
> Hi, yes, changing to [www.google.com](http://www.google.com) seems to be working, thank you! Now just to realize why doesn't ArduinoHttpClient library work, but I guess I have to check that with authors of the library.
> 
> Thank you again.

Hi @ibrcic, I guess I have the same problem, but changing URL don't help me. 
Just "+CME ERROR: operation not allowed", error 601... 
Please can you send here final code that 100% works?
Thank you in advance,
RevoltaAmadeus

ilgarbenli, 2023-12-05:
could you solve this https problem?

ilgarbenli, 2023-12-10:
> lease can you send here final code that 100% works?
> Thank you in advance,
> RevoltaAmadeus

I got "+CME ERROR: operation not allowed" too. I solved with using this library "https://github.com/govorox/SSLClient"