Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #162: Guide me to use 128x64 Monochrome LCD with u8g2 lib.

mashrurbd, 2021-09-09:
First of all, thanks for the library! I have been using TTGO T-Call for less than a month now and I am fond of its features!

I am using a 128x64 Monochrome LCD with u8g2 library.  The problem is that the display does not work all the time.

In their demo code, they mentioned a constructor:
```
// Please note that I changed the SPI, Correct me if I am wrong.
#define LCD_CS 33
#define LCD_SCLK 14 //25
#define LCD_MISO 12 //27
#define LCD_MOSI 13 //26

// SPIClass SPI1(HSPI);

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ LCD_SCLK, /* data=*/ LCD_MOSI, /* CS=*/ LCD_CS, /* reset=*/ 22);
```

In the last line, i.e.:
`U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ LCD_SCLK, /* data=*/ LCD_MOSI, /* CS=*/ LCD_CS, /* reset=*/ 22);`
the reset pin is set to GPIO: 22, can you suggest me another pin for this constructor? 

Also, I need to change these pins: 
```
#define LCD_SCLK 14 //25
#define LCD_MISO 12 //27
#define LCD_MOSI 13 //26
```
**Please reply with a correct constructor and pins, Thanks in advance!**

LilyGO, 2021-09-11:
You first need to know which model your OLED screen driver chip is.

mashrurbd, 2021-09-11:
I am using 128x64 LCD Monochrome display, not the OLED display. 
Please have a look at the photo:
[image]

closed 2021-10-15 by LilyGO

mashrurbd, 2022-07-08:
Please note that the controller model is: ST7920