Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #163: pushbutton on T-Call-SIM800C (with) AXP192

ClemensGruber, 2021-09-11:
[image]

LilyGO, 2021-09-13:
> What is the pushbutton on the C model next the SIM800C for, see picture? And how it is connected to the board. Can I use it as "common" push button, is it connected to an ESP pin?
> 
> On https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/ I found
> 
> > If you use the AXP192 power management chip, you need to press and hold the PWR button for 2 seconds to turn on the power supply. By default, the AXP192 will be powered on when the battery is inserted.
> 
> Do I have to switch on a feature for that? With a sketch based on the [TinyGSM example](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_TinyGSM) I can push the button short or 2 sec or longer with no effect. So it seems that I have to activate it somewhere -- or, in case I do not use the feature, I can use the button for my purpurse in case it is connected to the ESP?
> 
> [image]

> What is the pushbutton on the C model next the SIM800C for, see picture? And how it is connected to the board. Can I use it as "common" push button, is it connected to an ESP pin?
> 
> On https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/ I found
> 
> > If you use the AXP192 power management chip, you need to press and hold the PWR button for 2 seconds to turn on the power supply. By default, the AXP192 will be powered on when the battery is inserted.
> 
> Do I have to switch on a feature for that? With a sketch based on the [TinyGSM example](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_TinyGSM) I can push the button short or 2 sec or longer with no effect. So it seems that I have to activate it somewhere -- or, in case I do not use the feature, I can use the button for my purpurse in case it is connected to the ESP?
> 
> [image]

Hello, this button is not connected to ESP32.

closed 2021-09-24 by LilyGO

capedra, 2022-01-25:
@LilyGO sometimes the board does not power on when I plug in the battery or press the EN button, forcing me to press the PWR button for 2 seconds to power up. I don't know how I may solve it since the PWR button goes to one of the PMU pins.