Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #165: Upload not possible at 5 of 10 bought T-Call SIM800C, need replacement urgent, how to detect faulty boards

ClemensGruber, 2021-09-17:
I bought a longer time ago 2 - at this time relatively new - T-Call SIM800C directly at LilyGo via aliexpress. We found the newer model with the AXP usefull because we can separate the device from power and it can run on at powering without pressing an additional button, what is needed with the IP5306 version. 

We made some software for a bee scale and have tested things the last weeks for an workshop with 20 participants / scales / TTGO T-Calls. I ordered 10 pieces from a local store in Germany, Eckstein components, and also 10 pieces again from LilyGo's aliexpress store. 

The 10 T-Calls from Eckstein did arrive a time ago and I had planed to prepaire the T-Calls today for the workshop end of September. 

I run also in the "can not upload issue" as described as first FAQ on https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800, but all hints did not work! I could not upload any code on the device. After a lot of trying - btw. the first orderd T-Calls had never this problem. I tried the next one from the Eckstein batch. Also not working. The thired did the upload without any problem. 

So in the end I could uplad software to 5 of 10 orderd T-Calls from Eckstein components. 

[image]

I have the impression that the ESP is working one or two seconds and then switched off while the SIM800 tries to conect. 

After some T-Calls it is very clear what moduls are faulty and what are working: With a new module out of the box, in case the green status LED and the red SIM800L LEDs are light up / blinking it is likely that upload works also and all is fine. 

[image]

In case the green LED is flashing a second after startup and then is off and only the read modem LED is shining the chance is high that you got the faulty moduls!

On the faulty moduls reset button is also not working and you see no debug output on the serial monitor! 

[image]

I hope that the 10 modules I orderd from aliexpress are working and that I got a fast replacement for the buggy modules by TTGO! We need the modules fast for our workshop and it would not a good advertising that we have to say it is a 50:50 chance that you get a working module. 

Perhaps someone from TTGO can enlight us what the problem behind is. Is it a wrong initial software that makes the AXP not working correctly or is it a hardware bug?

amotl, 2021-09-18:
Dear Clemens,

did you also try all other workarounds found on the internet for this specific problem? There are a bunch of suggestions around and many of them worked for others. The most prominent ones are:

1. Use a better USB cable.
2. Use a baud rate of 115200 or 57600.
3. Hold down the programming button for the whole flashing process.
4. Don't write anything to "STDOUT" within your `void setup()`.
5. Add capacitors.
6. Check solder points.

With kind regards,
Andreas.

P.S.: To support you with more details about those workarounds, please inspect those resources:

- https://github.com/espressif/arduino-esp32/issues/485
- https://github.com/espressif/arduino-esp32/issues/1253
- https://esp32.com/viewtopic.php?t=18123
- https://esp32.com/viewtopic.php?t=23140
- https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/
- https://github.com/cnlohr/espusb/issues/8
- https://community.platformio.org/t/a-fatal-error-occurred-invalid-head-of-packet/10233
- https://forum.sparkfun.com/viewtopic.php?t=47871
- https://forum.arduino.cc/t/unable-to-upload-esp32-code-invalid-head-of-packet-0x1b/577925
- https://www.esp8266.com/viewtopic.php?t=442
- https://rntlab.com/question/esp32-cam-a-fatal-error-occurred-failed-to-connect-to-esp32/
- https://georgik.rocks/esp32-erase_flash-failed-with-invalid-head-of-packet/
- https://forum.pycom.io/topic/1735/failed-to-connect-to-esp32-invalid-head-of-packet-x08
- https://stackoverflow.com/questions/61691037/cant-upload-my-code-to-esp32-and-not-able-able-to-erase-flash
- https://stackoverflow.com/questions/66855393/fail-arduino-sketch-upload-to-wemos-lolin-d32-timeout-or-invalid-head-of-pac

ClemensGruber, 2021-09-18:
Today I checkt all 10 boards again. From the 5 yesterday not working boards is one working today. One from the yesterday ok boards had initial problems to be re-programmed. Ofter some trials it worked again. Nothing special changed just repeated trying. 

I would really have an comment from LilyGo about the presumable reason for this instability and non-functionality. 

Thanks Andreas for your time but I fear we have other reasons here, but I checked this also: 

>    Use a better USB cable.

I have tried at least 3 different USB cable, all cable working with other ESP boards and all cable working with at least 5 of the 10 bought T-Calls. So in this case it is not the cable! ;-) 

>    Use a baud rate of 115200 or 57600.

I get no serial connection on this faulty boards so this could not be the problem. But I treid 115200 also with no success. 

>    Hold down the programming button for the whole flashing process.

You are refering to the grounding GPIO0 as described in the FAQ (1. Open the serial monitor, 2. Short IO0 to GND, 3. > Press the RESET button ... serial should output `waiting for download`) I tried this and got the ouput waiting for download on one board, but without further success, see the error messages in my initial posting 

>    Don't write anything to "STDOUT" within your void setup().

I can not upload any sketch so I can not write somethin in setup() :-)

>    Add capacitors.

That is the one thing I did not yet, but it's working with the other boards so I think it's not the reason. 

>    Check solder points.

Modules are out of the box, I solderd only some screw terminals and header pins on one module that is not working. I double checked all solder points for solder bridges but all is ok for my soldering. Don't know how good LilyGo's quality check is.

amotl, 2021-09-18:
>> Hold down the programming button for the whole flashing process.
>
> You are referring to the grounding GPIO0 as described in the FAQ?

Maybe. Can you try this procedure **without** opening the serial monitor? I believe this might block the communication.

>> Don't write anything to "STDOUT" within your void setup().
>
> I cannot upload any sketch so I can not write something in setup() :-)

I see. In one of the links, there is also a note about erasing the flash completely by using `idf.py`. Maybe this helps.

ClemensGruber, 2021-09-18:
Of cause I did close the serial monitor. With no success. But I got - for testing - the output `waiting for download` just with one board, maybe the one that is working now.

LilyGO, 2021-09-18:
Hello，
Let me first help you solve the problem that you cannot download. Please open the serial port monitoring software, after connecting the device, double-click the RTS on the serial port monitoring software, that is, reverse it to see what information is printed on the serial port.

ClemensGruber, 2021-09-18:
I'm runnig the Arduino IDE in version 1.8.15 on a Win10 machine. After pluggin in the T-Call the serial port is detectet and shows up under toos > port. Serial moitor is set to 115200 there is not output on the console also after pressing reset on the T-Call or connect GND to GPIO0 and press and hold or press only without holding the reset button. 

On Arduino's seial monitor there is no RTS button to double-click. I tried also Putty but there is also no RTS to click. What software are you using for this task?

ClemensGruber, 2021-09-18:
I have now tried a Raspberry Pi, so a complet other system, but with the same problems. I also tried out some serial monitor programs on Linux and on Windows but non had an RTS button to click or an other interface to force RTS signals. Any help appreciated.

ClemensGruber, 2021-09-20:
Any hint @LilyGO? otherwise I will return the T-Calls to your reseller and hope that the replacement boards working.

LilyGO, 2021-09-24:
Have you tried to upload the firmware using flash_download_tool to see if it can be successfully uploaded?

ClemensGruber, 2021-09-24:
I have returned the non-working boards to your reseller due to time constrains, we need working board for a planed workshop. I will get the replacement boards today as announced by the postal service. I will check the boards and in case in this batch is also a non working I will try a binary upload via the ESP Flash Download Tool. In case someone is searching: https://www.espressif.com/en/support/download/other-tools

closed 2021-10-15 by LilyGO

crbble, 2021-10-31:
Just for reference, I had the same problem without any solution. Eckstein replaced one board without problems.

LilyGO, 2021-11-01:
Hello, can you take a video to show me how the board can't upload firmware?

ClemensGruber, 2021-11-18:
Because 3 of the 5 former working T-Calls from Eckstein did quit the job now also, I have the chance to try an upload via Espressif's ESP32 Flash Download Tool instead of the Arduino IDE as mentioned above. The 3 boards worked before soldering and have now the same behaviour as described above at the fresh unpacked one. One is working the most time but unreliable, one is working 20-30 seconds, seems to have something to do with heat ... 

It's also not possible to upload code with Espressif's tool! Serial port is detected but there is no further connection established. The tool is stalling at the sync process: 

[image]

Message after stopping: 

[image]

In the console you get after stopping this: 

```
test offset :  65536 0x10000
case ok
..........................................................................................................................................................................
[2021-11-18 11:04:52,222][ESP8266Loader_spi[1]][espDownloader.py][line:422][ERROR]: Chip sync error
[2021-11-18 11:04:52,222][ESP8266Loader_spi[1]][espDownloader.py][line:582][ERROR]: ESP32 Chip sync error esp_sync_blocking.
```

Good thing: All T-Calls directly from TTGO via aliexpress ordered are working and the replacement T-Calls from Eckstein also. Unfortunately Eckstein nor an other European reseller has T-Calls to sell. So I will go to aliexpress again and hope that shipping is not too slow!