Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #167: Unable to register network after inserting SIM pin

pcarmo, 2021-09-24:
Hi,

I've created a code, using the arduino framework and the TinyGsm library to unlock a SIM800l modem with a SIM card and register the module on the network.
I have tested the same code on 3 different SIM800L IP5306 20190610 boards, but I've only been able to register in the network with one of the boards.
The other ones I receive the SMS ready and Call Ready messages, but when I send the command "AT+CSQ=?", i keep getting "0,0" response.

I've [read somewhere](https://forum.arduino.cc/t/gsm-module-sim800l-no-signal/479829/10) that the SIM800L requires a large amount of current (2A) to register a network, but besides USB I also tried other power source, linked directly to the 5V pin. 
I assume the board would be able to support the required amount of power to register the network, since it is also prepared to work with a battery.

Does somebody know what else could I try?

Thanks in advance.

Here are the set of commands that I send (>>) and the responses I receive (<<). The symbol (...) means some time has been passed between 2 responses:
>> AT
<<
<< OK
>> AT+CFUN=0
<<
<< +CPIN: NOT READY
<<
<< OK
>> AT+CFUN=1,1
<<
<< OK
>> AT+CPIN?
<<
<< +CPIN: SIM PIN
<< 
<< OK
>> AT+CPIN="5029"
<<
<< OK
<<
<< +CPIN: READY
<<
<< +CGREG: 2
<<
<< +CGREG: 3
<<
<< Call Ready
<<
<< SMS Ready
(...)
<< 
<< +CREG: 0
>> AT+CSQ
<<
<< +CSQ: 0,0
<<
<< OK

LilyGO, 2021-10-07:
+CREG: 0,3,  and 3 are registration denied.Most likely it's the SIM card.Make sure your card is in order

LilyGO, 2021-10-15:
Does your card support GSM ?

LilyGO, 2021-10-18:
Can't you get a signal properly using a factory configured antenna?

LilyGO, 2021-10-20:
turnOffNetlight ();It's just used to turn off the network light.Network registration and use are not affected

eljefe40, 2021-10-20:
well i would assume that to be true but .. explain to me 2 things...
2 boards brand new.. 

1. no red LED activity as would be expected with sim800l.
commenting line out or using the complimentary turnOnNetlight () does NOT result in ANY activity on the net light..
never seen the red LED ever illuminate and I have several other modem boards from other manufacturers with the same SIM800l part that have working net leds

AND most importantly:

 with that SINGLE change (commenting line out), it will  NOT connect to GPRS. 
if this is unrelated, then there is something WRONG with BOTH of these NEW T-call devices.

makes no sense to me but.... it follows this pattern consistently.. 
just telling you my observation.

zoe9216, 2023-01-05:
Hi,
My SIM800L was working fine for two months with a separate power source (5V, 3A buck converter) and connected to a NodeMCU-32S. Now it doesn't connect to network anymore, LED keeps blinking every second continuously and returns +CREG: 0,3. I have hard reset it via RST pin and it returns CREG: 0,2. But after a while switches back to 0,3.
I checked with my country's telecommunication authority and my device is registered just fine. 

Kindly tell me how to fix this?

michaelradhuber, 2024-03-06:
I have had the same issue right now, broken/bad Antenna connection was the cause. Notwithstanding that CSQ showed good signal of 22! Best, Mike

well0nez, 2024-07-02:
> n/bad Antenna connection was the cause. Notwithstanding that CSQ showed good
 hey michael, how did you fix it?
 i have many modules and the same problem, signal is good but the connection is so poor...

michaelradhuber, 2024-07-03:
First improve your signal with a good antenna, check all connections and soldering. Switch the module, many are faults.