Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #173: SIM800C stopped responding to AT commands

crbble, 2021-11-02:
Hi, I had a running AXP192 20200609 board that registered to the network etc. Unfortunately while trying a few things the SIM800C stopped responding to AT commands. 

I already tried the ResetModem sketch including the StreamDebugger, but this requires that communication to the SIM800C is working. 

Interestingly, I still receive RDY, +CFUN 1 and CPIN: NOT INSERTED from the modem once the red light turns on, but when I issue AT or something else I do not receive anything.

This is example output from the ResetModem sketch. First time with 115200 baud (this was working before), afterwards with 9600. 

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1216
> ho 0 tail 12 room 4
> load:0x40078000,len:10944
> load:0x40080400,len:6388
> entry 0x400806b4
> VBUS:5093.20 mV 0.00 mA ,BATTERY: 0.00
> 
> Testing Modem Response...
> 
> ****
> 255
> ⸮
> RDY
> 
> +CFUN: 1
> 
> +CPIN: NOT INSERTED
> 
> ****
> 
> ***********************************************************
>  Failed to connect to the modem! Check the baud and try again.
> ***********************************************************
> 

I also tried all other baudrates and some autobaud things, like sending AAAAAT for autobaud. A second board works with the same sketch. Also interesting, the other board outputs some values for "VBUS:5093.20 mV 0.00 mA ,BATTERY: 0.00" instead of zero.

Is there any connection on the board I can check? I do not have a signal analyzer, so unfortunately I cannot hook up to the TX/RX pins. I already checked VEXT, this is 2.8 V.

Thanks.

LilyGO, 2021-11-03:
Hello, is the module status light steady on?

crbble, 2021-11-03:
Hi, the ESP was working normally. The SIM800C status was steady on, the net light was blinking in one of the usual state frequencies (before initialization I think), basically the SIM800C looked normal. 

I solved the issue by following the upgrade firmware guide. I flashed the UpgradeFirmware sketch to the ESP, used mtkdownload on linux (I ignored that the simcom.ee page certificate has expired) and flashed ROM_VIVA from that spurious archive 1418B07SIM800C24 that I downloaded from onedrive. What maybe also helped was resetting the internal filesystem on the SIM800C (this is an option to mtkdownload). I was playing with the upload SSL certificate sketch, maybe that broke something.

closed 2021-11-03 by crbble

samrepublicc, 2024-09-03:
same problem it is not workingwith any board of me but working with only ttl adapter