Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #177: Brownout when start wifi and modem together

garudaonekh, 2021-11-20:
Hi, 

I got strange Brownout exception when I call begin Wifi and start the modem together.

If i donot start wifi, the modem can work and connect GPRS, SMS, call normally

LilyGO, 2021-11-22:
Hello, which board do you use ?  Can you send in your source code ?

garudaonekh, 2021-11-22:
T-Call V1.4. SIM800L_IP5306_VERSION_20200811

LilyGO, 2021-11-22:
I tried WiFi. The equipment is operating normally.

garudaonekh, 2021-11-27:
It happened sometimes if I starts Wifi and GSM modem. I don't have any clear pattern. But I use only GSM modem, never have this brownout issue.

ClemensGruber, 2021-12-06:
How is the board connecte to a power source? Could it be an insufficient power source?

garudaonekh, 2021-12-07:
> How is the board connecte to a power source? Could it be an insufficient power source?

main cause of brownout is power issue but I do not think this is my case. sometimes I restart several time and it start working normally

ClemensGruber, 2021-12-07:
I had several boards that worked normally with some sensor readings or also with additional wifi. But only a too thin extension cable of the JST LiPo leads to a non functional GSM modem. 

Can you assign the brownout to special functions or behaviours? E.g. transmitting data via modem or initialize the GSM connection? Or starting WiFi while GSM is connected?

The GSM alone needs 2 A peak, some old USB standards are below this 2 A. So I would first try to connect to an other USB power source or a fully charged LiPo battery and see if the problem occures still. Perhaps you may check the max current of your Mac's USB port in case it is a older model.

ibrcic, 2021-12-07:
Hi ClemensGruber, I have exactly the same problem as shown in the video in the link you posted. My German is a bit rusty so could you please tell me how did you solve the problem? Do I need to solder the battery directly to the board or use shorter wires? Currently I am using wires about 1 meter long, maybe that could be a problem?

ClemensGruber, 2021-12-09:
@ibrcic in the **non working** video I had different power management devices between LiPo and T-Call: 

**LiPo / solarpanel > CN3065 solar charger > TPL5110-Timer > T-Call**

A 2000 mA/h LiPo and a 2 W solar panel was connecte to a CN3065 breakout board, this is a dedicated solar charger. The charger was connected to a timer to save more energy than possible with the T-Call only. I have around 10 of this settings running. But could not manage to get around 8 working the same way. Changing single components did not work. So it must be a problem with adding factors. 

I have also two JST cable soldered together as connection to the solar charger (without the TPL5110 breakout) but the same effect, some are working some are not. 

So my advice would be use a standard LiPo and connect it direct to the JST plug on the T-Call. In case you are using a 18650 you will need a battery holder. The items with spring connections are reported as bad for high current. Use the one with "spring plate" instead, for a picture see https://de.aliexpress.com/item/33048934797.html

ibrcic, 2021-12-09:
I just tried soldering thicker wire directly to JST plug on the T-Call and other end directly to the battery terminals but it didn't help. The modem simply wont connect when using the battery. What is strange to me is that it is working normally through USB port that is connected to the PC and that can't provide more than 1000mA if I remember correctly.

proximus323, 2021-12-12:
T-Call V1.4. SIM800L_IP5306_VERSION_20200811 not working on battery a have 3 board. 
Can some on tell how to power up on battery ?

proximus323, 2021-12-12:
This board sucks, I put 4.2 V from the power supply, it start working but for about 30 second voltage for 5V PIN got to 2.25 V and 3.3V PIN is now 0V, ESP32 not wake up from deep sleep, ith only work on USB port.
This can NOT be used in any projects.

garudaonekh, 2021-12-13:
> This board sucks, I put 4.2 V from the power supply, it start working but for about 30 second voltage for 5V PIN got to 2.25 V and 3.3V PIN is now 0V, ESP32 not wake up from deep sleep, ith only work on USB port. This can NOT be used in any projects.

it's working again and the voltage is back to 3.3v after you reconnect the power or reset the board?

proximus323, 2021-12-13:
Power boost for IP5306 !!!

I spend cople of hours to get this work.

bool setupPMU()
{
bool en = true;
Wire.begin(I2C_SDA, I2C_SCL);
Wire.beginTransmission(IP5306_ADDR);
Wire.write(IP5306_REG_SYS_CTL0);
if (en)
{
Wire.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
}
else
{
Wire.write(0x35); // 0x37 is default reg value
}
return Wire.endTransmission() == 0;
}