Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #178: Failed at AT+CREG?

garudaonekh, 2021-11-22:
Hi,
My SIM800L V1.4 SIM800L IP5306 20200811 failed at AT+CREG?  (Network registration) after several days. SMS worked(I saw AT Command received on the terminal) and the RING also worked. But GPRS not worked due to this AT+CREG? failure. It showing this AT Command(AT+CREG?).

I change the SIM to another SIM800L of the same model and it worked. Same TTGO model, same code, same computers, same SIM, same USB cable.

LilyGO, 2021-11-23:
Send the AT + CPIN? Check whether a SIM card exists

garudaonekh, 2021-11-24:
It look like it doesn't response to AT command but it still can response to incoming call and sms. i can see AT response "RING" when I make a call to the installed SIM.

LilyGO, 2021-11-25:
Use this [example ]( https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_AT_Debug)to debug the AT directive

41linea45, 2021-12-22:
Hello

I have same problem. I tried Arduino_AT_Debug example but i cant send ant AT command to module.
Module always says AT+CREG? .

But it can accept CALL ans sms . no network and internet. 
I set apn.
Everyting is good. Yesterday I could reach network properly.

garudaonekh, 2021-12-24:
> Hello
> 
> I have same problem. I tried Arduino_AT_Debug example but i cant send ant AT command to module. Module always says AT+CREG? .
> 
> But it can accept CALL ans sms . no network and internet. I set apn. Everyting is good. Yesterday I could reach network properly.

I don't think it's related with APN. I think it's related with SIM800 module hardware issue or Telecom issue. Not with your code.

41linea45, 2021-12-24:
> I don't think it's related with APN. I think it's related with SIM800 module hardware issue or Telecom issue. Not with your code.

Thank you for reply . I have TCall SIM800**C** too. I can work with it properly without issue.
How can i fix SIM800L module ? firmware reupload? 
when i read sim800 rx,tx external uart module, i can reading response.

garudaonekh, 2021-12-24:
The reason I said it's hardware issue because error only happened sometimes. I use T-Call Sim800L.
Sometimes I have encountered BrownOut issue which is said to be related with power issue.