Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #179: SD and SIM800L

irenelectronica, 2021-11-23:
I have conected sd card in (25,27,26,33) like the example. However when I try to write in SD and send data with GSM doesn't work. I'm always in this step Connecting to APN:....  and it doesn't connect. I realise that pines 26 and 27 also are used by sim800 .... What can I do?
Thanks

irenelectronica, 2021-11-24:
I have changed the sd pines to (25,0,2,34), by that way there are not pines used by sd and by sim800l.
I have achieve to do a cycle, but after, it did not work
Any ideas? Thanks

MKoroschetz, 2021-11-29:
> I have changed the sd pines to (25,0,2,34), by that way there are not pines used by sd and by sim800l. I have achieve to do a cycle, but after, it did not work Any ideas? Thanks

We are using the SD card as follows, and it is working fine with GSM transmission. 
`def init_sd(enabled=True, mount=True):
  ''' Activate SD Card reader using SPI interface
      tested on T-Call v.1.3, 1.4 board
      miso = Pin(19, Pin.INOUT, Pin.PULL_UP)
      mosi = Pin(15, Pin.INOUT, Pin.PULL_UP)
      slck = Pin(18, Pin.INOUT, Pin.PULL_UP)
      cs   = Pin(14, Pin.INOUT, Pin.PULL_UP) <- Pin(5) conflicting PIN with GSM modem!
      spihost = 2
      maxspeed = 16
  '''  
  if enabled:
    uos.sdconfig(uos.SDMODE_SPI, clk=18, mosi=15, miso=19, cs=14, spihost=2, maxspeed=16)
........`

We are using [loboris micropython ](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/gsm)

mlrochaa23, 2022-07-14:
@irenelectronica Did you find a solution for this problem? I'm having the same issue using CANBus Module. It also connect to ESP32 via SPI, like the SD Module.

MKoroschetz, 2022-07-14:
The solution is described in our post @mkoroschetz. It has been working very well since we adopted the described changes.

mlrochaa23, 2022-07-14:
Ok. I will try that. Thank you!

msentuerk, 2022-07-18:
You should be also careful which SD Library you are using. In Platform IO there are several SD Libraries. The only working one with custom SPI and specially for the ESP32 T-Call is this one "SD(esp32) @ 1.0.5". Here is also a working code.

`SPIClass sdSPI(VSPI);

File myFile;

void initSDCard() {

  sdSPI.begin(15, 13, 2, 12);

  Serial.print("Initializing SD card...");
 
  if (!SD.begin(12, sdSPI)) {
    Serial.println("initialization failed!");
    while (1);
  }
  Serial.println("initialization done.");

}`