Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #183: How to turn off GSM Module status LED

ibrcic, 2021-12-06:
On SIM800C AXP192 20200609 there is both Netlight led that lights up when there is network communication going on as well as Status LED that lights up when modem is powered on.

I understand that it is possible to turn off Netlight using `AT+CNETLIGHT=0` but I am unable to find AT command for turning off the status LED. Is it even possible or will I need to physically remove it from the board?

This is the LED that I am talking about:
[image]

ClemensGruber, 2021-12-20:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/e4b487a54f5f5c1baa82048a0941d2504e55e23b/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino#L159-L163

ibrcic, 2021-12-20:
That is Netlight, I am wondering about status light, which is a different thing unfortunately.

ClemensGruber, 2021-12-20:
The two last lines

https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/e4b487a54f5f5c1baa82048a0941d2504e55e23b/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino#L162-L163

ibrcic, 2021-12-20:
Ahh, that's unfortunate :/

closed 2022-01-14 by ibrcic