Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #184: Built-in LED indicators - external connection

rytisp, 2021-12-06:
Has anyone experienced how to make built-in LED indicators external? Through GPIO?

ibrcic, 2021-12-09:
Depends on version of the board and LEDs you are talking about.

There is regular LED on PIN 12 (on my board) that you can easily make external through PIN 12. There is also Netlight which shows activity of GSM modem, I'm not sure that you can make this one external other than soldering directly to the board. There is also modem status light that is always on if modem is on, I guess you could check if modem is working and use that to put some PIN to HIGH. There is also battery charging LED on my board, you can get charging current from the AXP and turn on some pin if it is > 0.

If you give me more details on board you have and LEDs you want to make external I can try and be more specific.

rytisp, 2021-12-11:
I'm currently using IP5306 20200811. 
Soldering directly to the LEDs themselves it's too dangerous as there are very small soldering pads. 
Power-on LED I'll manage something. Mostly interested in GSM activity LED.