Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #189: Some T-Call-SIM800C do not work below 3.9 V LiPo voltage

ClemensGruber, 2022-01-04:
I have around 30 T-Calls here and after #165 I am a bit upset to face the next problem with this unstable T-Call boards! 

Some boards can not establish a GPRS connection via the SIM800 module with a LiPo below 3.9 V voltage. The same LiPo, the same SIM card, the same code are working on an other T-Call. All T-Calls, working and non working are labeled as: 

```
20-6-9
T_CALL&sim800C-DS_V02
```

You can see data loss / less transmissions below 4.0 V LiPo voltage
[image]

For this issue see also https://community.hiveeyes.org/t/insuffiziente-stromversorgung-des-sim800-moduls/4263/3 [German content]

ClemensGruber, 2022-01-04:
The working module with voltage below 3.9 V is a T-Call I ordered a time ago, it has a different silkscreen at the ESP32 module as you can see on the left side. The not working module is on the right side, I ordered them 2021-11-18 via Lilygo's AliExpress shop:

[image]

ClemensGruber, 2022-01-05:
Here a ongoing test with the **older T-Call SIM800C (left side on the pic in the last posting)**. I connected the discharged LiPo with ~3.9 V to this board and could send around **1470 additional datasets** down **to 3.56 V** (measured by the AXP). 

[image]

While testing I had to change my software - you can see the time at the blue triangle - I had a very conservative threshold definde to prevent LiPo from deep discharge - and after setting this to a lower limit the T-Call could send data down to to 3.56 V!

That means that it was possible to send **1800 datasets with a LiPo form 4.2 to 3.5 V with one T-Call** vs. **400 datasets only** with a newer T-Call and a Lipo form 4.2 to 3.9 V. What was changed in the meantime? Why do the other T-Call stopp working at 3.9 V? Why is this change in hardware not reflected by the silkscren number or note? In case there was no hardware chang (I can not imagine) is there any setting in the AXP chip that is not overwritten by uploading new software and we can and should change? 

The older T-Call is really working good but what a mess with the newer / other one! Any suggestin from @LilyGO what I can do to get the modules working down to 3.5 V?

LilyGO, 2022-01-06:
Hi, what version does the old version mean?  Have a look at the picture of the product?

LilyGO, 2022-01-06:
How many boards now appear below 3.9VGSM working incorrectly.  What kind of electricity are you using now?  Will USB power supply work?

ClemensGruber, 2022-01-09:
[image]

I have spent a hour to test all 16 T-Call SIM800C boards I have laying around at my workbench. 

- all 16 working via SIM800 with USB-Power
- all 16 working via LiPo with ~4.1 V 
- **14 do not work via LiPo with ~3.8 V** 
- 2 are working via LiPo with ~3.8 V, but one T-Call only one time, so its more ore less 1 board with no problem and working with voltage below 3.9 V

[image]

The working board is the one on the left side in [posting #2](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/189#issuecomment-1004680787) in this thread. As far as I can reconstruct it is an "older" T-Call SIM800C, I have ordered Mar., 28th 2021 via LilyGo's AliExpress shop. 

The non working T-Calls with SIM800C are ordered Nov., 18th 2021 via LilyGo's AliExpress and some board from your reseller Eckstein, I have ordered September 2021 in Germany. 

For testing the 16 boards I have used a 2000 mAh LiPo from Eckstein, see https://eckstein-shop.de/LiPoAkkuLithium-IonPolymerBatterie32C7V2000mAhJST-PHConnector but I have tested some other LiPos (one from Adafruit, 2200 mAh and one from SODAQ, 1200 mAh) with the same outcome. I have also tested with a power bench supply, same issue! So the problem is not the LiPo! 

[image]

Have you anything changed in production / in the board design. Or can you imagine that some settings of the AXP are by default different at this different production runs? I just want to know: Can I something do against this miss-function via software or is it a hardware bug. 

We have bought some test boards before making the project running with your T-Calls and tested also discharging as you can see on my [comment from Jul., 2nd 2021](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/80#issuecomment-873001194). We have bought more boards with the expectation that all is running as tested before and stuck now with deployed sensor kits in the wilde we do not know how long they will work.

ibrcic, 2022-01-10:
I have exactly the same problem with latest AXP board, the only difference being that I am using Li-ion battery.

When powering the board from USB (which afaik is limited to 1A) everything works as expected. But when I power the board from battery that in theory can supply up to 10A peak current, I am unable to connect to the GSM network when voltage drops below about 4.0V or 4.1V. 

I was able to somewhat mitigate the problem by connecting 470uF capacitor to battery terminal which lets me use the battery untill about 3.7V, which is still not close to using the full capacity.

ClemensGruber, 2022-01-10:
I had a detailed look at the two boards the working one is the on the **left** side vs. the not working! I found some not populated parts at the working board! So on the working bord the parts are "missing" or not placed. 

[image]

I have sacrified one board and tried to desolder this parts but made some collateral damage, my skills and equipment is not sufficient for this small structures. 

Could this be the reason for the different behaviour? Are this parts ment to be placed or DNP parts and they should not be on the board?

ibrcic, 2022-01-10:
Definitely interesting that the board without some components is working correctly, and as far as I can tell they are both latest version with AXP192. It would be great if @LilyGO could shed some light into what are these components for?

ClemensGruber, 2022-01-12:
Any answer from @LilyGO about this issue?

joseantoniovalero, 2022-01-13:
The  470uF capacitor to battery, in positive terminal ?

ibrcic, 2022-01-13:
I connected the capacitor to positive and negative battery terminal on the board.

ibrcic, 2022-02-03:
Any news on this issue? Seems like it is impacting a lot of users.

ClemensGruber, 2022-02-07:
@LilyGO we are still waiting for an answer.

komkritc, 2022-02-12:
Same problem !

LilyGO, 2022-02-14:
HI, we are planning to switch to other power circuit designs.

ibrcic, 2022-02-14:
@LilyGO that's great to hear! Will there be another revision of the board so that we know when to order it? I am waiting with ordering of several boards until this issue is resolved.

realbeeit, 2022-02-14:
@LilyGO when we can expect revision for order?

ClemensGruber, 2022-04-07:
You can find a project documentation under https://community.hiveeyes.org/t/openhive-esp32-ttgo-t-call-sim800c/4202 and corresponding code via https://github.com/ClemensGruber/open-hive_bee-monitoring/blob/master/open-hive_esp32_ttgo-t-call-sim800c.ino but with the current hardware from TTGO I could not recommand the TTGO T-Call!

andrea-993, 2022-04-28:
Is there any way to mitigate the problem?
So far what i have found out is that:
-SIM card with better operator coverage does not cause this issue (less power is used to communicate)
-disable wifi if you need to use the sim800c helps a bit
-solder a 200uF (or bigger) low ESR capacitor between GND and DVDD4V (look on schematics)
still, the problem may appear from time to time
if anyone has better ideas please share

ClemensGruber, 2023-03-10:
> HI, we are planning to switch to other power circuit designs.

Have you done this now? Are now ordered T-Calls usable below 3,9 V LiPo voltage or do you still sale the design I got January 2022?

ClemensGruber, 2023-10-30:
@LilyGO Are any of the now (10/2023) available T-Calls fixed and work up to 3.5 V? Seems the overview on https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800 is not up to date because in the shop I see a SIM800**L(!)** module with AXP192 https://de.aliexpress.com/item/4000959701330.html and a V1.4 labeled version with SIM800H https://de.aliexpress.com/item/33045221960.html, but I think last one is with IP5606 and off by connection the LiPo to the boad.

Rob58329, 2024-04-05:
Also see https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/87 (470uF capacitor)