Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #194: LilyGo-T-CALL-SIM800L power supply problem

Tirititelu, 2022-02-01:
Hello, I want to realize a project using LilyGO-T-Call-800L (IP5306 20190610) consisting of a captive portal to which users access to send an SMS by pressing a button on the page of the same. To do this I want to power the LilyGo with a Li-Ion battery, which is also charged with a photovoltaic plate. This will be done using the MT3608 and TP4056 modules. I have done similar projects with other microcontrollers like arduino MKR and similar, no problem, but in this case the LilyGo does not work, it does not even create the WIFI access point.

I do not go into the details of the connection between the modules for now because I think the problem comes from another side. I have noticed that when I power the LilyGO only with the battery and put in series a multimeter to measure the intensity, it does not work, it does not turn on (after pressing the reset button as indicated in the documentation). I have tried with several different multimeters with the same result, can anyone know the reason? I have the impression that when connecting any of the modules mentioned above happens the same as when trying to average the intensity and therefore does not work, but I can not think what it can be and how to solve it.

Thank you very much.

ClemensGruber, 2022-02-07:
You may google "burden voltage" or have a look at https://edadocs.software.keysight.com/kkbopen/34970a-what-is-burden-voltage-588262398.html or Andreas' explanation https://www.youtube.com/watch?v=LUB8RWzzLWc -- in short your measurement drops power and leads to a non functional SIM800 or ESP32. In the most cases you ESP will run but WiFi is a problem and of cause the SIM800 because it needas up to 2 A peak current.

sajalbhargava, 2022-09-19:
i faced a similar issue while measuring current from my fluke 179 multimeter.
so what i did was connected both red and black connectors  of multimeter together and then pressed the reset button 
remove after 5 sec it will work properly.

Rob58329, 2024-04-05:
Also see https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/87.
(Adding anything between the lipo and the T-Call-SIM800 is problematic as it drops the current available to the IP5305 which is running on it's limits anyway.)