Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #195: ESP32 sim800l starts with battery only after pressing the reset button

tom2323tom2323, 2022-02-08:
Hi there,

When I use the ESP32sim800l (KENEW Lilygo TTGO T-Call V1.4 ESP32 Wireless Module SIM Antenna SIM Card SIM800L Module (CP2104 V1.4 Version))
with the USB-C it starts right away.

But when I use the battery power connection, it only comes on after pressing the reset button. But I would like the ESP32sim800l to go directly to the battery power connection (without pressing the reset button). Is that possible?

vahooz, 2022-02-13:
I have the same problem.

LilyGO, 2022-02-14:
This is a feature of IP5306

ClemensGruber, 2022-02-18:
See the FAQ section at https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800

> 4. Why doesn’t the board turn on when the lithium battery is connected?
> 
>     If you use the IP5306 version and use a lithium battery to supply power, you must press the reset button to turn on the board.

This is not the case at with boards with AXP192 as PMU.

Ribisl, 2022-03-13:
Any progressions here? I'm facing the same problem, maybe you can fix it with i2c?

tavdog, 2022-05-02:
This sucks.

smurfolan, 2022-06-16:
I have a similar case. I power the module from the Type-C input using [this](https://www.trust.com/en/product/21149-primo-powerbank-10-000-mah) power bank.  The specification of this battery says that its composition is **Lithium-ion cell**. I use the 5V2.1A output of the battery and the module never starts.

Once I turn the battery on, there's a very strange click-click noise coming from the battery itself (probably some kind of internal protection) and after 5-10 seconds it turns off from itself.