Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #199: TT GO T CALL PMU SIM800L AXP192 20200327 power issues

htopalov, 2022-03-27:
Hello, I've made a simple gps tracker with this controller and Neo 6m gps module sending data to server which is working fine when connected to the usb. But when I power the board with 3.7v 1700mAh lipo battery it can't send gps data to server. Actually it does send it but only from time to time.....1 out of 10 maybe even less. According to specifications 3.7v battery should be enough to power the board correctly. I'm even using this peace of code from examples to set the power boost and keep working on battery when usb is disconnected:
`bool setPowerBoostKeepOn(int en){
  I2CPower.beginTransmission(IP5306_ADDR);
  I2CPower.write(IP5306_REG_SYS_CTL0);
  if (en) {
    I2CPower.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  } else {
    I2CPower.write(0x35); // 0x37 is default reg value
  }
  return I2CPower.endTransmission() == 0;
}`

htopalov, 2022-03-28:
For anyone who is here for the same problem.... I came with a solution taken from other open issues here and over the net and tested(one time for now) on my boards. Just solder 470uF 10V capacitor to the battery(parallel connection) . Capacitor compensates voltage drop when SIM800L is triggered. If not voltage drop increases and SIM800L restarts or it is stuck and can't perform request over internet neither send SMS... well just about anything. Keep testing it...

closed 2022-03-28 by htopalov

ibrcic, 2022-03-30:
For anyone else stumbling on this issue. This is a problem with hardware design of power circuit of the device. LilyGo acknowledged this issue and said that they are working on new revision of the board that should be done in a few weeks (at least that's what they told me last time I asked customer support on AliExpress)

htopalov, 2022-03-31:
After some test with my devices and wiring I've managed to to get it work for 4 hours before battery drops to 3.8V and Sim module drops dead(can't connect to network). 4 hours is ridiculously small amount of time. I've managed to test and in ma case battery drops at about 0.1V/h which is ok. So the last thing that I can come up with is cheap boost dc dc converter with enough amps to supply sim module and board. I've order two types from aliexpress and when they come I will test again and report here. P.S. with converter I should be able to get lower values as much as 2.7-3V(lowest battery voltage before integrated battery protection shuts it down) and still working stable but time will tell....

htopalov, 2022-04-07:
Reporting another issue here for anyone who is struggling with the same problem....power issues go further when changing sim antenna....If using bigger antennna (8dBi instead of 5dBi) and if powered from same battery from my first post description, problem get even bigger.... Tested in area with low gsm signal coverage..result - can't connect properly to network... two gps receivers with two different gsm providers... After doing this test several times, I tried to come up with reason for the problem...
Well I took my powerbank in order to provide external power source and everything started working fine for 8dBi antenna device. Second device was tested along with the described an it was working fine with orignal antenna provided with board and without any external power source. So keep in mind... antenna and power... keep coding and searching

evisvasiu, 2022-05-20:
I am still struggling with this issue. Adding the capacitor improved a little bit because the board still won't turn on with the battery at 3.8-3.9V. I think it will just be better to put an external charger module and power the board from the 5V input.