Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #201: pin reading in loop

gunter72, 2022-04-03:
Hello everyone, I am creating a project and I have configured pins 15, 18 and 19 in INPUT_PULLUP with internal resistor
I have a problem, some cards give random readings on these pins.
Sometimes they seem to loop 0-1-0-1 ...
I use these versions:
v1.3 20190610
v1.4 20200811

thanks for the support.

LilyGO, 2022-04-06:
Hello. Can you send me your code?

gunter72, 2022-04-09:
of course, to what address?

LilyGO, 2022-04-12:
Just send it right here

gunter72, 2022-04-27:
unfortunately I can't post the code, it's not all mine

gunter72, 2025-05-09:
I solved my problem, I was using pins 34 and 35 as RX and TX for the GPS module, I moved them to pins 32 and 33, now the board is much more stable and I no longer have the loop problem.

closed 2025-05-09 by gunter72