Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #203: Deep Sleep on Battery won't wake up

johncs1609, 2022-04-20:
I have a problem that when the ESP32 goes into deep sleep, it cuts or turns off the boost and cuts the ESP32 power and does not return from deep sleep anymore! is there any way to solve this?

LilyGO, 2022-04-22:
Refer to the [example](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino),  which has periodic sleep wake up at the end

Jerr4y, 2023-04-23:
> I have a problem that when the ESP32 goes into deep sleep, it cuts or turns off the boost and cuts the ESP32 power and does not return from deep sleep anymore! is there any way to solve this?

how did you solve it, can you speek english? if yes we can start a discord call it would help me A LOT

Rob58329, 2024-04-05:
See https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/87