Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #205: Error when setting Root Certificate on SIM800L

Uf0Lab, 2022-04-22:
I really need some help on 2 questions:

1. I'm trying to connect a T-Call SIM800L (IP5306 20190610) to AWS IoT, which requires an SSL authentication through a **Root Certificate**, a **Client Certificate** and a **Private Key**.

I'm able to successfully create the .crt files, write on them and set the Client Certificate through the AT Command:

    modem.sendAT(GF("+SSLSETCERT=C:\User\clientcert.crt"));

But when I try to set the RootCA like this:

    modem.sendAT(GF("+SSLSETROOT=C:\User\rootca.crt,1188"));

the GSM module returns "ERROR".

[This](https://microchip.ua/simcom/2G/Application%20Notes/SIM800%20Series_SSL_Application%20Note_V1.05.pdf) is the documentation I'm using as a reference (Page 13).

In the "Reference" section of the command "AT+SSLSETROOT", it's written "**The files to be imported must be binary encoded**". I'm not sure whether this is what's causing the issue, but this confused me a little. At first I simply wrote on the file the string format of the certificate, but I've also tried to change the extension to .der (which technically is the binary encoded format for certificates). I've tried to write on the file the hexdump version of the string format, and tried all the other avaiable extensions (.crt, .cer, .pem, .p12).

I've also tried to update the firmware following the procedure [here](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/doc/How%20to%20update%20firmware.md).

I saw on other discussions that the SIM800L **does not support TLS 1.2**, but on page 6 of the SIM800 SSL documentation it's reported that "SIM800 series support SSL2.0, SSL3.0, TLS1.0 and **TLS1.2**."

On top of that, the command "AT+CIPSSL=1" works fine since it returns "OK".

I also read [this](https://github.com/vshymanskyy/TinyGSM/issues/29#issuecomment-328802556).

I'm attaching the function that sets up the certificates.

```
void setCertificates() {

    modem.getModemInfo();

    modem.sendAT(GF("+FSCREATE=C:\\User\\rootca.crt"));
    modem.waitResponse();
    modem.sendAT(GF("+FSCREATE=C:\\User\\clientcert.crt"));
    modem.waitResponse();

    char rootcertific[1188];
    strcpy(rootcertific,rootCA);

    modem.sendAT(GF("+FSWRITE=C:\\User\\rootca.crt,0,1188,1"));
    modem.waitResponse(">");
    SerialAT.print(rootcertific);
    modem.waitResponse();

    delay(1000 / portTICK_PERIOD_MS);

    char clientcertific[2903];
    strcpy(clientcertific,certificate_pem_crt);

    modem.sendAT(GF("+FSWRITE=C:\\User\\clientcert.crt,0,2900,2"));
    modem.waitResponse(">");
    Serial1.print(strcat(clientcertific,private_pem_key));
    modem.waitResponse();

    delay(1000 / portTICK_PERIOD_MS);

    modem.sendAT(GF("+FSREAD=C:\\User\\rootca.crt,0,1188,1"));
    modem.waitResponse();
    modem.waitResponse();

    modem.sendAT(GF("+FSREAD=C:\\User\\clientcert.crt,0,2900,1"));
    modem.waitResponse();
    modem.waitResponse();

    modem.sendAT(GF("+SSLSETROOT=C:\\User\\rootca.crt,1188"));
    modem.waitResponse();

    modem.sendAT(GF("+SSLSETCERT=C:\\User\\clientcert.crt"));
    modem.waitResponse();
}
```

2. On page 11 of the SIM800 SSL documentation, the command "AT+SSLSETCERT" is described as "Client Client Certificate File with Private Key". Does this mean that I somehow have to write **both the Client Cert and the Private Key on the same file**?

As you might have noticed in the function, I used "strcat(clientcertific,private_pem_key)" and it returns no error, but I'm not sure whether this is the correct way to do this.

If anyone can help me out on this one I'd be infinitely grateful. It's been keeping me stuck for almost a month now.

Thank you!

aleksanderLisiecki, 2022-06-09:
Hi! I am struggling with the same problem, did you find any solution?

drpetersen, 2023-05-27:
Almost one year later … so am I. I have a SIM800L R14.18 (Revision:1418B05SIM800L24 as per `ATI` and `AT+CMGR` commands). I can set a client certificate, but not a root certificate.

What is more, the module does not even seem to recognize the `AT+SSLSETROOT` command. Whereas `AT+SSLSETCERT=?` responds with `+SSLSETCERT: 100,32`, `AT+SSLSETROOT=?` yields a `+CME ERROR: …`.

My conclusion is that this has simply not been implemented on the SIM800x devices. The [application note](https://microchip.ua/simcom/2G/Application%20Notes/SIM800%20Series_SSL_Application%20Note_V1.05.pdf) referenced by @WhiteUfoDev above is version 1.05. The [manufacturer's website](https://simcom.ee/modules/gsm-gprs/sim800/) (you have to accept the expired certificate, which does not even match the domain if you want to view this … :roll_eyes:) for the SIM800x product links to a list of [Application Notes](https://simcom.ee/documents/?dir=SIM800x) among which **version 1.02** of the SSL application note. At the bottom of page 9 of [that document](https://simcom.ee/documents/SIM800x/SIM800%20Series_SSL_Application%20Note_V1.02.pdf), it says "Currently, we just support SSL Client function."

So I guess, we're out of luck and will stay that way for want of a newer firmware … :disappointed:

drpetersen, 2023-05-27:
And for anyone else running into this also trying to get this to work with an ESP32: The way to go for me was using the [SSLClientESP32](https://github.com/alkonosst/SSLClientESP32) library instead. :sweat_smile: Works well together with both TinyGSM and the Arduino HttpClient, even in the same sketch and switching dynamically between WiFi and GSM.

DREU007, 2025-04-24:
> And for anyone else running into this also trying to get this to work with an ESP32: The way to go for me was using the [SSLClientESP32](https://github.com/alkonosst/SSLClientESP32) library instead. 😅 Works well together with both TinyGSM and the Arduino HttpClient, even in the same sketch and switching dynamically between WiFi and GSM.

Thx. Worked with SIM800C.