Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #208: Performing HTTPS +CME ERROR: operation not allowed

emil-malmgaard-rasmussen, 2022-05-06:
Hello,

Ihave followed your example code, and tried to connect to my api using HTTPS, I'm getting AT+HTTPSSL=1 OK 
So I'm assuming that the connection should succeed, however, that is not the case.

this is the following that seems to be going wrong, I really have no idea how to proceed from here:
```
01:41:24.879 -> GPRS connected
01:41:24.879 -> Performing HTTPS GET request... AT+HTTPTERM
01:41:24.913 -> 
01:41:24.913 -> +CME ERROR: operation not allowed
01:41:24.913 -> [24193] +HTTPTERM
01:41:24.913 -> AT+SAPBR=0,1
01:41:24.913 -> 
01:41:24.913 -> OK
01:41:25.912 -> AT+SAPBR=3,1,"Contype","GPRS"
01:41:25.912 -> 
01:41:25.912 -> OK
01:41:25.912 -> AT+SAPBR=1,1
01:41:25.912 -> 
01:41:25.912 -> OK
01:41:25.912 -> AT+SAPBR=2,1
01:41:25.912 -> 
01:41:25.912 -> +SAPBR: 1,1,"10.161.128.115"
01:41:25.912 -> 
01:41:25.912 -> OK
01:41:27.930 -> AT+HTTPINIT
01:41:27.930 -> 
01:41:27.930 -> OK
01:41:27.930 -> AT+HTTPPARA="CID",1
01:41:27.930 -> 
01:41:27.930 -> OK
01:41:27.930 -> AT+HTTPPARA="URL","www.api.tooltracker.dk"
01:41:27.930 -> 
01:41:27.930 -> OK
01:41:27.930 -> AT+HTTPPARA="REDIR",1
01:41:27.930 -> 
01:41:27.930 -> OK
01:41:27.930 -> AT+HTTPSSL=1
01:41:27.930 -> 
01:41:27.930 -> OK
01:41:27.930 -> AT+HTTPACTION=0
01:41:27.930 -> 
01:41:27.930 -> OK
01:41:37.927 -> AT+HTTPREAD
01:41:37.963 -> 
01:41:37.963 -> +HTTPACTION: 0,603,0
01:41:37.963 -> 
01:41:37.963 -> OK
01:41:37.963 -> AT+HTTPTERM
01:41:37.963 -> 
01:41:37.963 -> OK
01:41:37.963 -> AT+SAPBR=0,1
01:41:37.963 -> 
01:41:37.963 -> OK
01:41:37.963 -> AT+CIPSHUT
01:41:38.397 -> 
01:41:38.397 -> SHUT OK
01:41:38.397 -> AT+CGATT=0
01:41:38.614 -> 
01:41:38.614 -> OK
01:41:38.614 -> GPRS disconnected
```

I have tried with URLS such as: Baido and Google.com. With Baido it was the same result with Google I got:
```
01:54:06.648 -> +HTTPREAD: 14803
01:54:06.648 -> <!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="da"><head><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><meta content="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop="image"><title>Google</title><script nonce="k1q-RgtqlkBBVDYbJwZYjg">(function(){window.google={kEI:'F7V1YujQLZmWjgbM0KPoCA',kEXPI:'0,1302536,56873,6059,206,4804,926,1390,383,246,5,1354,4013,1238,1122515,1197780,621,380089,16115,28684,17572,4858,1362,12313,4751,12835,4998,13228,3847,3600,592,6430,22741,2370,4304,1279,2742,149,1103,840,6297,109,4011,2023,1777,520,14670,3227,2845,7,17450,8102,7666,552,4465,13142,3,576,6608,13975,4,1528,2304,32112,2658,7357,13658,4437,16786,5803,2554,4094,4052,3,3541,1,42154,2,14022,1931,998,11187,11623,5679,3401,28742,4568,6255,20803,2618,1246,5841,12137,2831,4332,6089,1395,1959,23017,1669,1,436,8155,6401,181,323,476,12,2437,12231,7341,2622,29,13733,1878,1308,1431,124,1882,2559,597,23,5101,2552,489,2716,3501,4,450,163,514,3,208,1916,1490,1510,214,1494,6436,356,1493,2738,239,1526,132,2,,19,158,450,157,754,3122,5429805,100,61,8796584,1326,3,1877,1,2562,1,748,141,795,6164,6,2,169,13,19,46,5,39,644,29,2,2,1,2,1,2,2,7,4,1,2,2,2,2,2,2,1052,1,1,158,3,2,2,2,2,2,4,2,3,3,3,1,1,1,1,62,21,5,9,22,14,2,58,19,2,40,1,10,3,2,2,8,2,75,23950681,4042143,1964,298',kBL:'53LM'};google.sn='webhp';google.kHL='da';})();(function(){
01:54:06.822 -> var f=this||self;var ar b;a&&(!a.getAttribute||!(b=a.getAttribute("eid")));)a=a.parentNode;return b||h}funct(!a.getAttribute||!(b=a.getAttribute("leid")));)a=a.parentNode;return b}
01:54:06.822 -> function n(a,search("&ei=")||(e="&ei="+l(d),-1===b.search("&lei=")&&(d=m(d))&&(e+="&lei="+d));d="";!"&cshid=")&&"slh"!==a&&(d="&cshid="+f._cshid);c=c||"/"+(g||"gen_204")+"?atyp=i&ct="+a+)+d;/^http:/i.test(c)&&"https:"===window.location.protocol&&(google.ml&&google.ml(Erro"");return c};h=google.kEI;google.getEI=l;google.getLEI=m;google.ml=function(){return b,c,d,g){if(c=n(a,b,c,d,g)){a=new Image;var e=k.length;k[e]=a;a.onerror=a.onload=a.ona;a.src=c}};google.logUrl=n;}).call(this);(function(){
01:54:06.892 -> google.y={};google.sy=[];google..id;else{do c=Math.random();while(google.y[c])}google.y[c]=[a,b];return!1};google.sx=f;google.lm=[];google.plm=function(a){google.lm.push.apply(google.lm,a)};google.lq=[];oogle.lq.push([[a],b,c])};google.loadAll=function(a,b){google.lq.push([a,b])};google.b}).call(this);google.f={};(function(){
01:54:06.929 -> document.documentElement.addEventListener("sub.target){var c=a.getAttribute("data-submitfalse");a="1"===c||"q"===c&&!a.elements.q.veventDefault(),b.stopPropagation())},!0);document.documentElement.addEventListener("cr(a=b.target;a&&a!==document.documentElement;a=a.parentElement)if("A"===a.tagName){a=href");break a}a=!1}a&&b.preventDefault()},!0);}).call(this);</script><style>#gbar,op:1px !important;}#gbar{height:22px}#guser{padding-bottom:7px !important;text-align:x solid #c9d7f1;font-size:1px}.gbh{height:0;position:absolute;top:24px;width:100%}@mein-right:.5em;vertical-align:top}#gbar{float:left}}a.gb1,a.gb4{text-decoration:underor:#00c !important}.gbi .gb4{color:#dd8e27 !important}.gbf .gb4{color:#900 !importan.h{font-family:arial,sans-serif}body{margin:0;overflow-y:scroll}#gog{padding:3px 8pxtd{line-height:17px}form{margin-bottom:20px}.h{color:#1558d6}em{font-weight:bold;fonx;width:496px}.gsfi,.lst{font:18px arial,sans-serif}.gsfs{font:17px arial,sans-serif:inline-block;margin:3px 0 4px;margin-left:4px}input{font-family:inherit}body{backg4b11a8;text-decoration:none}a:hover,a:active{text-decoration:underline}.fl a{color:}.sblc{padding-top:5px}.sblc a{display:block;margin:2px 0;margin-left:13px;font-size;border:solid 1px;border-color:#dadce0 #70757a #70757a #dadce0;height:30px}.lsbb{dinline-block;margin:0 12px}.lsb{background:url(/images/nav_logo229.png) 0 -261px repsor:pointer;height:30px;margin:0;outline:0;font:15px arial,sans-serif;vertical-aliadce0}.lst:focus{outline:none}.tiah{width:458px}</style><script nonce="k1q-RgtqlkBBgoogle.erd={jsr:1,bv:1579,de:true};
01:54:07.107 -> var f=this||self;var g,h=null!=(g=f.mei)?g:1,m,google.erd,u=r.jsr;google.ml=function(a,b,d,k,c){c=void 0===c?2:c;b&&(q=a&&a.messal(a,c,d),null;if(0>u){window.console&&console.error(a,d);if(-2===u)throw a;b=!1}eling script"===a.message||p>=h&&!k?!1:!0;if(!b)return null;p++;d=d||{};var e=c;c=encp=i&ei="+c(google.kEI);google.kEXPI&&(b+="&jexpid="+c(google.kEXPI));b+="&srcpg="ver="+c(r.bv)+("&jsel="+e);e=a.lineNumber;void 0!==e&&(b+="&line="+
01:54:07.175 -> e);var l=a.fill===window.location.href&&(e=document.documentElement.outerHTML.split("\n")[e],b+No script found.")));for(var t in d)b+="&",b+=c(t),b+="=",b+=c(d[t]);b=b+"&emsg="+sst="+c(a.stack||"N/A");12288<=b.length&&(b=b.substr(0,12288));a=b;k||google.log(function(a,b,d,k,c){q!==a&&(a=c instanceof Error?c:Error(a),void 0===d||"lineNumb==b||"fileName"in a||(a.fileName=b),google.ml(a,!1,void 0,!1,"SyntaxError"===a.namstring(0,11)?2:0));q=null;n&&p>=h&&(window.onerror=null)};})();</script></head><b"k1q-RgtqlkBBVDYbJwZYjg">(function(){var src='/images/nav_logo229.png';var iesg=ion(){window.n && window.n();if (document.images){new Image().src=src;}
01:54:07.245 -> if (!iesg);document.gbqf&&document.gbqf.q.focus();}
01:54:07.245 -> }
01:54:07.245 -> })();</script><div id="mngb"><div id/b> <a class=gb1 href="https://www.google.dk/imghp?hl=da&tab=wi">Billeder</a> <le.dk/maps?hl=da&tab=wl">Maps</a> <a class=gb1 href="https://play.google.com/?hlhref="https://www.youtube.com/?gl=DK&tab=w1">YouTube</a> <a class=gb1 href="httper</a> <a class=gb1 href="https://mail.google.com/mail/?tab=wm">Gmail</a> <a clacom/?tab=wo">Drev</a> <a class=gb1 style="text-decoration:none" href="https://ww?tab=wh"><u>Mere</u> &raquo;</a></nobr></div><div id=guser width=100%><nobr><spa=gbf class=gbf></span><span id=gbe></span><a href="http://www.google.dk/history/k</a> | <a  href="/preferences?hl=da" class=gb4>Indstillinger</a> | <a target=_.google.com/ServiceLogin?hl=da&passive=true&continue=https://www.google.com/&ecr></div><div class=gbh style=left:0></div><div class=gbh style=right:0></div></dd"><div id="lga"><img alt="Google" height="92" src="/images/branding/googlelogoor_272x92dp.png" style="padding:28px 0 14px" width="272" id="hplogo"><br><br></><table cellpadding="0" cellspacing="0"><tr valign="top"><td width="25%">&nbspinput name="ie" value="ISO-8859-1" type="hidden"><input value="da" name="hl" type="hidden" value="hp"><input name="biw" type="hidden"><input name="bih" type=ht:32px;margin:4px 0"><div style="position:relative;zoom:1"><input class="lst tx 0 6px;vertical-align:top;color:#000;padding-right:38px"  [... cut ...] rkt⸮jer\x22,\x22psrc\x22:\x22Denne s⸮gni3d\\\x22/history\\\x22\\u003EWebhistorik\\u003C/a\\u003E\x22,\x22psrl\x2 billede\x22,\x22srch\x22:\x22Google-s⸮gning\x22},\x22ovr\x22:{},\x22pq\x222:[],\x22sbas\x22:\x220 3px 8px 0 rgba(0,0,0,0.2),0 0 0 1px rgba(0,0,0,0.0x22scd\x22:10,\x22stok\x22:\x22GbLqQZ6-7wCJwLSeKSwNDeVt8OY\x22,\x22uhde\x22</script>        </body></
```

But I'm not sure if it didn't just default to http and get the html from there.

Been stuck for some time, help is much appreciated!

ilgarbenli, 2023-12-05:
could you solve this problem, I get same error messages "+CME ERROR: operation not allowed", anyone help?

PeteDouma, 2025-03-10:
I get the same error