Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #212: I can't send SMS with Lilygo TCALL SIM8OOL 1.4 version

FlasterX, 2022-07-06:
Hello I have bought a lilygo TCALL SIM800L 1.4 version but the SMS send fonction doesn't work with the exemple on github.
Does anyone have a solution ?

[image]

I have used this code : 

> 
> // Please select the corresponding model
> 
> #define SIM800L_IP5306_VERSION_20190610
> //#define SIM800L_AXP192_VERSION_20200327
> //#define SIM800C_AXP192_VERSION_20200609
> //#define SIM800L_IP5306_VERSION_20200811
> 
> //const char simPIN[]   = "1234";
> 
> // Define the serial console for debug prints, if needed
> #define DUMP_AT_COMMANDS
> #define TINY_GSM_DEBUG          SerialMon
> 
> #include "utilities.h"
> 
> // Set serial for debug console (to the Serial Monitor, default speed 115200)
> #define SerialMon Serial
> // Set serial for AT commands (to the module)
> #define SerialAT  Serial1
> 
> // Configure TinyGSM library
> #define TINY_GSM_MODEM_SIM800          // Modem is SIM800
> #define TINY_GSM_RX_BUFFER      1024   // Set RX buffer to 1Kb
> 
> #include "TinyGsmClient.h"
> 
> #ifdef DUMP_AT_COMMANDS
> #include "StreamDebugger.h"
> StreamDebugger debugger(SerialAT, SerialMon);
> TinyGsm modem(debugger);
> #else
> TinyGsm modem(SerialAT);
> #endif
> 
> // Set phone numbers, if you want to test SMS and Calls
> #define SMS_TARGET  "+33"
> 
> 
> 
> 
> 
> void setup()
> {
>     // Set console baud rate
>     SerialMon.begin(115200);
> 
>     delay(10);
> 
>     // Start power management
>     if (setupPMU() == false) {
>         Serial.println("Setting power error");
>     }
> 
>     // Some start operations
>     setupModem();
> 
>     // Set GSM module baud rate and UART pins
>     SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
> 
>     delay(6000);
>     //if (strlen(simPIN) && modem.getSimStatus() != 3 ) {
>    // modem.simUnlock(simPIN);
>   //}
> }
> 
> void loop()
> {
>     // Restart takes quite some time
>     // To skip it, call init() instead of restart()
>     SerialMon.println("Initializing modem...");
>     modem.restart();
> 
> String modemInfo = modem.getModemInfo();
>     SerialMon.print("Modem: ");
>     SerialMon.println(modemInfo);
> 
>     delay(10000);
> 
>     String imei = modem.getIMEI();
>     DBG("IMEI:", imei);
> 
>     bool  res = modem.sendSMS(SMS_TARGET, String("Défaut de commande Digesteur 1 ") + imei);
>     DBG("SMS:", res ? "OK" : "fail");
> 
>     // Do nothing forevermore
>     while (true) {
>         modem.maintain();
>     }
> }

pilot1981, 2023-02-16:
Hi,
you understood which are RX and TX pin connected between ESP32 and SIM800L?

amiiit, 2024-03-04:
I have also bought what seems to be the same board and I have no success with the demo code.It seems like it has nothing to read from the SerialAT.  I believe TX and RX should be set according to sketch attached to this issue, which is 26&27.  Anyways, even flipping these didn't work. 

I wonder, the gpio's 26&27 this refers to the ESP right, not to the SIM Module?

Does anyone have a working example for this board? It seems to be like a weird mix of v1.4 PCB, but with the old SIM800L module, perhaps this messed up the wiring?

pilot1981, 2024-03-05:
it's my working code for ESPHOME:

esphome:
  name: combinatoregsm

ota:
  safe_mode: true
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
#  level: DEBUG
#  baud_rate: 0 # disable uart logger on esp 8266

# Enable Home Assistant API
api:
  services:
    - service: send_sms
      variables:
        recipient: string
        message: string
      then:
        - sim800l.send_sms:
            recipient: !lambda 'return recipient;'
            message: !lambda 'return message;'
    - service: dial
      variables:
        recipient: string
      then:
        - sim800l.dial:
            recipient: !lambda 'return recipient;'
    - service: connect
      then:
        - sim800l.connect
    - service: disconnect
      then:
        - sim800l.disconnect
    - service: send_ussd
      variables:
        ussdCode: string
      then:
        - sim800l.send_ussd:
            ussd: !lambda 'return ussdCode;'

text_sensor:
  - platform: template
    id: sms_sender
    name: "Sms Sender"
  - platform: template
    id: sms_message
    name: "Sms Message"
  - platform: template
    id: caller_id_text_sensor
    name: "Caller ID"
  - platform: template
    id: ussd_message
    name: "Ussd Code"

binary_sensor:
  - platform: sim800l
    registered:
      name: "Sim800L Registered"

uart:
#  debug:
  baud_rate: 9600
  tx_pin: 27
  rx_pin: 26

switch:
  - platform: gpio
    id: pwrkey
    pin: 4
    restore_mode: always_off
  - platform: gpio
    id: rstpin
    pin: 5
    restore_mode: always_on
  - platform: gpio
    id: pwron
    pin: 23
    restore_mode: always_on

sensor:
  - platform: sim800l
    rssi:
      name: "Sim800L RSSI"
  - platform: wifi_signal
    name: "WiFi Signal Sensor Combinatore GSM"
    update_interval: 60s

sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
  on_incoming_call:
    - lambda: |-
        id(caller_id_text_sensor).publish_state(caller_id);
  on_call_connected:
    - logger.log:
        format: Call connected
  on_call_disconnected:
    - logger.log:
        format: Call disconnected
  on_ussd_received:
    - lambda: |-
        id(ussd_message).publish_state(ussd);

ArashIranfar, 2024-04-24:
I don't know the details but this setup function solved my problem: 

```
void setup()
{
    SerialMon.begin(9600);

    delay(10);

    if (setupPMU() == false) {
        Serial.println("Setting power error");
    }
    setupModem();

    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);

    SerialMon.println("Initializing modem...");
    modem.restart();

    delay(10000);

      SerialAT.println("AT+CSMP=17,167,0,8"); 
      updateSerial();
      delay(500);

      String imei = modem.getIMEI();
      DBG("IMEI:", imei);

      bool  res = modem.sendSMS(SMS_TARGET, String("Hello from ") + imei);
      DBG("SMS:", res ? "OK" : "fail");
}
```

The key line is : _** SerialAT.println("AT+CSMP=17,167,0,8"); **_
Apparently it should be the default but with some network providers you need to send it again.

vAugagneur, 2024-06-12:
@ArashIranfar i have the same probleme, can you give the code of updateSerial() please ? is it personal code or somthing coming from utilities.h ?

Best Regards

ArashIranfar, 2024-06-12:
I made no changes to the updateSerial() function. It is the normal one you see in every Sim800 example:

void updateSerial(){
  while (Serial.available()){
    SerialAT.write(Serial.read());//Forward what Serial received to Software Serial Port
  }
  while (SerialAT.available()){
    Serial.write(SerialAT.read());//Forward what Software Serial received to Serial Port
  }
}

Also check the local forums to see if the module is compatible with your local service provider. In my region there are 3 SPs and only one requires the following line:

 SerialAT.println("AT+CSMP=17,167,0,8");