Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #216: Need help: I connected the battery to the esp32 and it does not work properly.

xBLACKICEx, 2022-08-14:
Need help: I connected the battery to the esp32 and it does not work properly.

Questions

* battery work must first be connected to usb c to
* Unplug the usb c can only work for 30 seconds~1min

[battry is EEMB 3.7 2000mAh(7.4Wh)](https://jp.eemb.com/public/Download/Rechargeable-Lithium-Battery/Li-Polymer-Battery/Standard-Version/LP103454.pdf)

Battery voltage is 4.09V when tested with a multimeter

Battery directly connected to esp32 outflow voltage is 2.25V~2.30V

Connect to usb c and pull out quickly, the battery can keep the voltage at 4.09v for a short time（30~60s）

Continuously connected to usb, the measured voltage is 4.15V

**The code I used to test** See if the led can light up
```c++
void setup() {
  pinMode(12, OUTPUT);
  digitalWrite(12, HIGH);
}

void loop() {
}
```

**I measured the voltage here on the board:**
[image]

LilyGO, 2022-08-15:
Hello，Can you tell me the version? silkscreen on board

xBLACKICEx, 2022-08-15:
> Hello，Can you tell me the version? silkscreen on board

Sorry I couldn't find the version on the board, so I took two photos

[image]
[image]

LilyGO, 2022-08-15:
Can you run this [example ](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_TinyGSM) normally?

xBLACKICEx, 2022-08-15:
> Can you run this [example ](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_TinyGSM) normally?

i think yes is print out 
[image]
in serial

LilyGO, 2022-08-16:
Can run the example normally

closed 2022-08-16 by LilyGO