Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #218: Connecting GT-7U GPS module

4Heisenberg, 2022-09-07:
Hi,

I am trying to connect a GPS module to the serial port. But I can´t even initialize the serial port. gpsSerial.available() returns always "0".

Is there any sample code for using the serial port (TX/RX) on the board?

#include "HardwareSerial.h";

#define GPS_BAUD 9600

// serial connection to the GPS device
HardwareSerial gpsSerial(2);

void setup()
{
    Serial.begin(9600);
    Serial.println("Setting up GPS");
    gpsSerial.begin(GPS_BAUD);
    Serial.println("GPS Status");
    Serial.println (gpsSerial.available());
}

void loop()
{

    // Output raw GPS data to the serial monitor
    while (gpsSerial.available() > 0) {
        Serial.write(gpsSerial.read());
    }
}

lewisxhe, 2022-09-08:
You just need to call Serial2.begin(YOUR_BAUDRATE, SERIAL_8N1, YOUR_GPS_RX_PIN, YOUR_TX_PIN) in setup

lewisxhe, 2022-09-08:
Here is simple example
```
#include "Arduino.h";

#define GPS_BAUD 9600
const uint8_t YOUR_GPS_RX_PIN = 14;
const uint8_t YOUR_GPS_TX_PIN = 15;

void setup()
{
    Serial.begin(9600);
    Serial.println("Setting up GPS");
    Serial2.begin(GPS_BAUD, SERIAL_8N1, YOUR_GPS_RX_PIN, YOUR_GPS_TX_PIN) 
    Serial.println("GPS Status");
    Serial.println (Serial2.available());
}

void loop()
{

// Output raw GPS data to the serial monitor
    while (Serial2.available() > 0) {
        Serial.write(Serial2.read());
    }
}
```