Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #219: LilyGo SIM800L IP5306 20200811 server connection problem

felipecookh, 2022-09-14:
Hi, I'm using LilyGo SIM800L IP5306 20200811.
In my code I want to establish connection client.connect(server, port) with server vsh.pp.ua (same as very common example) and it fails (never goes beyond that part of the code, onboard red LED gets stuck in client.conect).
I've checked that the GPRS connection is established using modem.gprsConnect(apn, gprsUser, gprsPass).
I've even checked in a different code that modem.getGSMDateTime(DATE_FULL) and SMS sending both work OK. I'm also using modem.waitForNetwork() as an additional measure.
I've checked using power with USB and/or LiPo battery. Same result in both cases, everything works except client.connect(server, port)
I've also used the example Arduino_ResetModem.ino in case that was the problem, but made no difference.

Any thoughts? Code below. Regards!

```
// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

#include "BluetoothSerial.h"
#include <Wire.h>
#include <TinyGsmClient.h>
#include <ESP32Time.h>

#include "WiFi.h"
#include "driver/adc.h"
#include <esp_wifi.h>
#include <esp_bt.h>

// Check if Bluetooth configs are enabled
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

// Bluetooth Serial object
BluetoothSerial SerialBT;

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWRKEY         4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT  Serial1

// Define the serial console for debug prints, if needed
//#define DUMP_AT_COMMANDS

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

#define IP5306_ADDR          0x75
#define IP5306_REG_SYS_CTL0  0x00

const uint64_t uS_TO_S_FACTOR = 1000000;  /* Conversion factor for micro seconds to seconds */
const uint64_t TIME_TO_SLEEP  = 5;        /* Time ESP32 will go to sleep (in seconds) */

// GPRS credentials
const char apn[]      = "bam.entelpcs.cl"; // APN other option is imovil.entelpcs.cl
const char gprsUser[] = "entelpcs"; // GPRS User
const char gprsPass[] = "entelpcs"; // GPRS Password

// Server details
const char server[] = "vsh.pp.ua";
const char resource[] = "/TinyGSM/logo.txt";
const int port = 80;                            // server port number

// Keep this API Key value to be compatible with the PHP code provided in the project page.
// If you change the apiKeyValue value, the PHP file /post-data.php also needs to have the same key
String apiKeyValue = "tPmAT5Ab3j7F9";

// TinyGSM Client for Internet connection
TinyGsmClient client(modem);

void setup() {

  // Keep power when running from battery
  bool isOk = setPowerBoostKeepOn(1);

  // Set modem pins
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_PWRKEY, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);

  delay(6000);

  digitalWrite(MODEM_RST, HIGH);

  // Turn on the Modem power first
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Pull down PWRKEY for more than 1 second according to manual requirements
  digitalWrite(MODEM_PWRKEY, HIGH);
  delay(100);
  digitalWrite(MODEM_PWRKEY, LOW);
  delay(1000);
  digitalWrite(MODEM_PWRKEY, HIGH);

  delay(6000);

  // Restart SIM800 module, it takes quite some time
  // To skip it, call init() instead of restart()
  modem.restart();
  // use modem.init() if you don't need the complete restart

  delay(10000);

  modem.waitForNetwork();

  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    while(1){};
  }
  
  if (!client.connect(server, port)) {
    while(1){};
  }

  // Close client and disconnect
  client.stop();
  modem.gprsDisconnect();
    
  // Pull down PWRKEY for more than 1 second according to manual requirements
  digitalWrite(MODEM_PWRKEY, HIGH);
  delay(100);
  digitalWrite(MODEM_PWRKEY, LOW);
  delay(1000);
  digitalWrite(MODEM_PWRKEY, HIGH);

  digitalWrite(MODEM_POWER_ON, LOW);

  delay(6000);

  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
  btStop();

  adc_power_off();
  esp_wifi_stop();
  esp_bt_controller_disable();

  esp_sleep_enable_timer_wakeup(uint64_t(TIME_TO_SLEEP * uS_TO_S_FACTOR));
  esp_deep_sleep_start();
}

void loop() {
}

bool setPowerBoostKeepOn(int en) {
  Wire.begin(I2C_SDA, I2C_SCL);
  Wire.beginTransmission(IP5306_ADDR);
  Wire.write(IP5306_REG_SYS_CTL0);
  if (en) {
    Wire.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  } else {
    Wire.write(0x35); // 0x37 is default reg value
  }
  return Wire.endTransmission() == 0;
}
```

lfdominguez-servinfo, 2023-02-27:
I'm presenting the same problem... in other board with a SIM800 I can use the HTTPClient that use the CIP method for direct TCP connection... but on this LiLyGO I get stuck on `AT+CIPSTART`, even before that I tried to use a CSQ to check signal quality and return nothing. But the GPRS was connected OK.

lfdominguez-servinfo, 2023-02-27:
Checked and found exactly where the SIM800 shutdown, is after i connect to APN, I dont know why (Using TinyGSM).

```c
_modem->gprsConnect(gConfig.apn.c_str(), gConfig.gprsUser.c_str(), gConfig.gprsPass.c_str())
```

After that line the SIM800 not respond any command. The last messages from SIM was:

```
...
OK
AT+CIFSR;E0

10.25.238.31

OK
```