Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #220: Upload firmware

fmnamado, 2022-09-23:
Hello,
I bought a TTGO Call board.
I think it was working ok.
I tried inserting a SIM card, powered on, the light flashed quickly and then slow.
I called the number and it was ringing.

I flash another firmware and I didn't understand at the time I would lost the original one.
Now the board doesn't do anything.
No led neither network initialization.

I tried updating the firmware as described here https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/master/doc/How%20to%20update%20firmware.md and it doesn't work. 
It never leaves the waiting state.
I tried to press the button, one time, several times, holding on, nothing.

The comms are working, checked with putty and see some cyclical text.

Some text while holding the on board button:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPReMQ}1M!}      ==T)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016▒▒Ғ▒Ҫ▒j

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
f+͡▒read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTC▒%E5}IMQ),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash rea*▒ɱ 1000
ets_main.c 371

So what it seems to me, is that I have lost the default firmware functionality, which enabled celular comms and built in fw update.
So I seem to have broken it.

Isn't there a way of directly burning the default firmware, complete, out of the box to restore the functionality?

Can you pinpoint me to a right direction to solve this?

Thank you very much.

fmnamado, 2022-09-23:
Another info:
I seem to be able to burn ESPHome firmware there.
I tried just now and it works.
Shouldn't I be able to burn default firmware also? Can you help me in doing it?
Thank you once again.

ridwanmullick, 2023-12-07:
hmm,  
extract this rar in ur dekstop 
https://drive.google.com/drive/folders/1k9O5FMFdCjmC-nE71rJQ1Ye88MYX38LK?usp=sharing
then select both the ino. files together and drag to the arduino ide app icon 
[image]
it will open then just "upload" (connect ur espmodule..  select the the module from that board selector and select  "esp32 dev module").   and for the esp32 libraries(additional settings for arduino ide) particularly of this controller is avvailable online (do this first) . hope it hepls them who are stuck.

PiotrBeeScale, 2024-01-17:
thx  ridwanmullick - resurection 
:)