Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #222: how to get  battery voltage with esp32-sim800L

xBLACKICEx, 2022-11-16:
how to get  battery voltage with esp32-sim800L ?

streem-ua, 2023-01-06:
Have you found any solution? I need to monitor battery voltage and also detect if USB connection is active or not.

lewisxhe, 2023-01-06:
If the PMU version is used, the PMU function can be used to directly obtain the voltage.

https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_adc/axp_adc.ino

If the power management of IP5306 is used, only the adc voltage of IO35 needs to be read. You can refer to here

https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/7ad1b2807a37b5a6ba33a4fb0efbe8466e96faa3/TFT_eSPI/examples/FactoryTest/FactoryTest.ino#L80

streem-ua, 2023-01-06:
@lewisxhe Thanks for reply! I think, that this is PMU version, because it doesn't power off, when I change power source from USB to LiPo. But I tried each one library separately. 

So, what I achieved using this [axp_adc.ino](https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_adc/axp_adc.ino)

```
const uint8_t i2c_sda = 21;
const uint8_t i2c_scl = 22;
...
AXP192/AXP202 ADC Test
Ooops, AXP202/AXP192 power chip detected ... Check your wiring!
```

And this [FactoryTest.ino#L80](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/7ad1b2807a37b5a6ba33a4fb0efbe8466e96faa3/TFT_eSPI/examples/FactoryTest/FactoryTest.ino#L80)

```
eFuse Vref:1107 mVVoltage :0.00V
...
Voltage :0.00V
```

Photo of my board:
[one](https://i.postimg.cc/N5PnCmrL/photo-2023-01-06-17-19-46.jpg)
[image]
[two](https://i.postimg.cc/N5PnCmrL/photo-2023-01-06-17-19-46.jpg)
[image]

Am I missing something?

lewisxhe, 2023-01-06:
You use IP5306, do you read IO35?
Here is a simple example

```
#include <Arduino.h>
#include "esp_adc_cal.h"

int vref =1100;

void setup()
{
    Serial.begin(115200);
    esp_adc_cal_characteristics_t adc_chars;
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);    //Check type of calibration value used to characterize ADC
    if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
        Serial.printf("eFuse Vref:%u mV", adc_chars.vref);
        vref = adc_chars.vref;
    } 
}

void loop()
{
    uint16_t v = analogRead(35);
    float battery_voltage = ((float)v / 4095.0) * 2.0 * 3.3 * (vref / 1000.0);
    String voltage = "Voltage :" + String(battery_voltage) + "V";
    Serial.println(voltage);
    delay(1000);
}
```

streem-ua, 2023-01-06:
@lewisxhe 
Yep! This shows up a right Voltage of battery, thanks!

`Voltage :4.35V`

Maybe you can help with one more thing, I need to monitor USB Type-C input, if it is available or not. (don't need to know USB voltage). I will be very grateful to you!

streem-ua, 2023-01-06:
For everyone struggling in the same situation, I figured out how to check power source, USB or Battery. 
So, I'll attach some pieces of code, hope you'll figure out how to use them.

```
#include <Wire.h>
#define IP5306_REG_READ_0 0x70
#define IP5306_GetPowerSource() ip5306_get_bits(IP5306_REG_READ_0, 3, 1)

uint8_t ip5306_get_bits(uint8_t reg, uint8_t index, uint8_t bits){
  int value = ip5306_get_reg(reg);
  if(value < 0){
    SerialMon.printf("ip5306_get_bits fail: 0x%02x\n", reg);
    return 0;
  }
  return (value >> index) & ((1 << bits)-1);
}

int ip5306_get_reg(uint8_t reg){
  Wire.beginTransmission(0x75);
  Wire.write(reg);
  if(Wire.endTransmission(false) == 0 && Wire.requestFrom(0x75, 1)){
    return Wire.read();
  }
  return -1;
}

void setup() {
  SerialMon.begin(115200);
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
...
  Wire.begin(21,22); // I2C_SDA, I2C_SCL
}

void loop() {
...
bool usb = IP5306_GetPowerSource();
String powerSource = usb?"USB":"BATTERY";
delay(3000);
...
}
```

Complete version of IP5306 registers, corresponding to power you can find [here](https://gist.github.com/me-no-dev/7702f08dd578de5efa47caf322250b57)

xBLACKICEx, 2023-01-07:
@lewisxhe
@streem-ua
Thank you guys. it look work for me