Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #228: problem to identify board and PINOUT connection

pilot1981, 2023-02-16:
Hi,
I buyed new board in attachment. I don't understand which version is this.

[image]

you can help me to understand version of boar and which pin are connected between ESP32 and SIM800L module? I have to configure them in ESPHOME firmware.

regards

lewisxhe, 2023-02-16:
This is the version of IP5306 power management. Pinout, look here

https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/blob/ef7ae1440150f5bc5a41f74bfd14447ea1c416c5/examples/Arduino_AT_Debug/utilities.h#L74

pilot1981, 2023-02-16:
It seems to have hardware problem on SIM800 module: if I call SIM in SIM800, ISP say it is in reachable…

How I can check it?

pilot1981, 2023-02-17:
How I can manage PIN and PUK is SIM?

To have all working fine I have to setup PIN to 0000?

lewisxhe, 2023-02-17:
Whether there is a pin code depends on whether your SIM card has set a password

pilot1981, 2023-02-17:
I have some chance to have it working in Italian GSM network?

lewisxhe, 2023-02-17:
If the operator GSM you use does not stop, it can be used. You just need to ensure that a SIM card without pin lock is connected to the board, and then connect the GSM antenna. Use AT+CGREG? Get the current card registration status to know whether it can be supported and used

pilot1981, 2023-02-17:
I’m using ESPHOME firmware

pilot1981, 2023-02-17:
I have to ma rain always:
PIN 4 to OFF
PIN 5 to ON

?

pilot1981, 2023-02-18:
I solved in this way:

PIN 4 always OFF
PIN 5 always ON
PIN 23 always ON

closed 2023-02-18 by pilot1981