Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #236: Urgent: Crashing when using ESP32-audioI2S

mkhan9047, 2023-05-09:
When calling the connecttoFs() or connecttohost(), that time the board is crashing. Tried with different pins but not working. Can I at least know some hints to customize or maybe fix and make a pull request? I am new to this library!

But I need to fix it urgently, Anyone please help? I am using a MAX98357A I2S Amplifier! Using below pins: 
// Digital I/O used
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Here's the log:
Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.
12:42:30.091 -> Core 1 register dump:
12:42:30.091 -> PC : 0x400907ee PS : 0x00060a33 A0 : 0x80092836 A1 : 0x3ffb1dc0
12:42:30.122 -> A2 : 0xb3ff8a0e A3 : 0x0000abab A4 : 0xb33fffff A5 : 0x00000001
12:42:30.122 -> A6 : 0x00060e20 A7 : 0x0000cdcd A8 : 0x0000abab A9 : 0x3ffb1ec0
12:42:30.122 -> A10 : 0x3f40b670 A11 : 0x3ffdc7b0 A12 : 0x3ffdc6c8 A13 : 0x3ffdbf2e
12:42:30.122 -> A14 : 0x3ffc6e20 A15 : 0x3ffb1ec0 SAR : 0x00000010 EXCCAUSE: 0x0000001d
12:42:30.154 -> EXCVADDR: 0xb3ff8a0e LBEG : 0x4008c14e LEND : 0x4008c159 LCOUNT : 0x00000000
12:42:30.154 ->
12:42:30.154 -> ELF file SHA256: 0000000000000000
12:42:30.154 ->
12:42:30.154 -> Backtrace: 0x400907ee:0x3ffb1dc0 0x40092833:0x3ffb1df0 0x40092e6d:0x3ffb1e10 0x40081fe1:0x3ffb1e30 0x40082218:0x3ffb1e50 0x4008165f:0x3ffb1e70 0x400ea7a3:0x3ffb1e90 0x400eaac9:0x3ffb1ec0 0x400ddd31:0x3ffb1f30 0x400ddda8:0x3ffb1f80 0x4010656e:0x3ffb1fb0 0x4008f7ca:0x3ffb1fd0

mkhan9047, 2023-05-09:
@lewisxhe @vshymanskyy  @LilyGO  @alokshaw  Please help 🙏🏽🙏🏽🙏🏽🙏🏽

lewisxhe, 2023-05-09:
I cannot see the problem from the log you provided. You need to use [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to translate the code. Then it will tell you exactly where the error is