Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #245: Powering Liligo Ttgo t-call

gpalmia, 2024-01-01:
I would like to power a Lilygo ttgo t-call controller (esp32 + sim800) with LiPo battery (type 18650).
Under the board there is a jst-ph 2mm connector but, if I connect, the LiPo battery to this connector the board is not powered.
However, I discovered that if I connect **together** the battery via jst and the USB port to the PC the board is powered and then **if I disconnect the USB socket** the board continues to be powered only by the battery. Is this normal behavior?
If so, how can you power the board without connecting to a PC, in particular if I use the deep sleep mode, during which the board stops being powered and then, when it wake up, powered again?

mashrurbd, 2024-02-05:
Please click on the EN button on the board.

gpalmia, 2024-04-01:
OK, if I press the EN button it work fine but if the code send the T-Call in sleep mode it does not awake.
Any advise?

Rob58329, 2024-04-05:
See https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/87